trouch/CreawsomeMod #5: Please Add bigger nozzle

Gaz83, 2019-04-27:
I wish to try your mod but I see the biggest Nozzle is 0.8mm. I am using a 1.2mm
Can you add this?

banolka, 2019-05-04:
I have same problem -I'm using 0.5 mm nozzle and cannot select it :(

jpcuzzourt, 2019-05-23:
I have been able to manually enable other nozzle sizes by modifying and duplicating and editing files in the *definitions* and *quality* subfolders.
There isn't a lot to it, but I haven't tested it out with the larger nozzles and actual prints. And some of the related calculated values probably need to have their relationships adjusted to work well. 
So I'll just second or third the request to expand the nozzle range available. 
I love this feature though. My solution previously was to create a new "printer" for each nozzle size.
This is so much more elegant!

fib3rh0ptics, 2019-05-24:
Seconded, thirded, and so on. The 8mm was dialed in quite nicely. I would love to scale it up to 1mm. This printer family has some rather large build areas that would benefit from it. 

Agreed on the elegance. The old way of handling nozzle changes felt so overcomplicated. This is great.

trouch, 2019-05-27:
0.3, 0.5 and 1.0 has been added in beta 3

Gaz83, 2019-05-28:
Please also add 1.2 nozzle I have a volcano X nozzle that I am desperate to use with this mod. 

Also, I noticed the 0.8mm nozzle is restricted to max height of 0.32. 
Shouldn't this be 80% of the nozzle diameter i.e. 0.64?

vinceswann, 2019-05-31:
Can I request 1.5mm nozzle support too please? I realise it's not exactly a common request - I drilled my own to make insanely sturdy prints at 1.2mm layer height.

closed 2020-03-21 by trouch