trouch/CreawsomeMod #7: Ender-3 isn't showing up when adding printer

compudaze, 2019-05-07:
Ender-3 isn't showing up when adding printer
[image]

compudaze, 2019-05-07:
Since I already had Cura 4 installed and an Ender-3 setup, I had to uninstall Cura and manually remove the install folder, reinstall, then copy the resources folder again. Now it shows up. Not sure why it didn't work with existing install.

madmoonfish, 2019-05-17:
I initially had the same issue, but i just removed my old ender-3 printer, restarted Cura and i was able to use the new ender-3 mod profile

peejayw, 2019-05-23:
I Had the same issue soremoved my old ender 3 printer and restarted, then the new? profile showed up. However it is just the old Ender profile, no mods.
If I add the Ender 10 profile, that is obviously the new one.
Any suggestions?

trouch, 2019-05-23:
There are probably some conflict with your existing setup and profiles, you may need to reset your personal settings but please use Cura backup first :-)

x1a999, 2019-05-23:
1st try these 2 steps and launch Cura (modded)
Navigate to C:\Users\[USERNAME]\AppData\Roaming\cura
Rename (or delete) the Cura version file (4.0 to 4.0.old)

[image]

If the above fails:
uninstalled and reinstalled Cura again.
Follow the above steps again (to be sure)

Hope this helps :)

XBrav, 2019-05-23:
@x1a999 has the right idea.

From what I've gathered, the printer id conflicts with an older imported Ender-3 profile. I can't recall what I did last night exactly, but you can modify the profile identifier in the Creawsome package, and it'll show up.

horninc, 2019-05-24:
The solution that I think is straight forward is the following:

Locate a file "creality_ender3.def.json" in your "\Ultimaker Cura 4.0\resources\definitions" and rename it to "_creality_ender3.def.json", it should "kick" the old profile and allow you to import it.

halversonea, 2019-05-24:
> I Had the same issue soremoved my old ender 3 printer and restarted, then the new? profile showed up. However it is just the old Ender profile, no mods.
> If I add the Ender 10 profile, that is obviously the new one.
> Any suggestions?

You need to delete your old material settings too and create new ones based on the defaults otherwise it won't import the new printer settings.

Regularest, 2019-05-24:
Im having the same settings, but i cant delete my old material settings, nor my print settings. I managed to get the ender 3 graphic to show up, but thats the only mod im getting from that.

jarjartee, 2019-05-25:
well - I appreciate the efforts here but so far the mod has made a total mess of my Cura4 and  settings (probably not helpful comment) on MacOS - going to try to reinstall Cura and 'reapply creawsome resources'. It did show up 1st time install but completely threw x/y offsets when I sent off to Octoprint although appears correct on build plate in Cura.

joelhaasnoot, 2019-05-25:
For Mac the folder to delete is `~/Library/Application Support/cura/4.0/` . So open a terminal window and do `rm -rf ~/Library/Application Support/cura`

trouch, 2019-05-25:
Please refer to that Ultimaker wiki page to cleanup cache on windows
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

WillieRipple, 2019-05-25:
very simple, add a generic printer, delete your old one, then add the new one you want. then you can delete the generic

b1naryVap0r, 2019-05-25:
had the same thing happen:

i had cura 3.6 installed, i installed cura 4.0 without uninstalling 3.6 and the option to add the ender 3 wasnt there after moving the resources folder over.

note that i did have my previously insalled ender 3 as an available printer so, i deleted it and attempted to add a printer but the option under creawesome was not there. 

i deleted all printers and closed cura 4, then uninstalled 3.6. i reopened cura 4 and still not an option, i deleted the resources folder and extracted again, didnt work, even redownloaded and nope.

uninstalled and reinstalled 4.0 then replaced the resources folder and boom, i can add the ender 3.

 i had a problem with the ender 3 communicating with cura after that but it is for a different reason, which i also had to modify when i installed 3.6, see here if you cant print via usb to ender 3 in cura <https://github.com/Ultimaker/Cura/issues/4196>

jarjartee, 2019-05-26:
Reinstalled everything from scratch - works now - great print results - thanks

AloftHobbies, 2019-05-26:
> The solution that I think is straight forward is the following:
> 
> Locate a file "creality_ender3.def.json" in your "\Ultimaker Cura 4.0\resources\definitions" and rename it to "_creality_ender3.def.json", it should "kick" the old profile and allow you to import it.
> 
> You will the need to go to Manage printers and remove the old Ender 3 profile you had and restart Cura.

I tried the other items on this page for my MAC, and this is the one that finally got things working. Thanks!

danielfmo, 2019-05-26:
I just need to remove the already existing default Ender 3 printer and add the new one.

horninc, 2019-05-27:
> I just need to remove the already existing default Ender 3 printer and add the new one.

Sure, but if you only have 1 printer set up in Cura, it does not allow you to remove it.

trouch, 2019-05-27:
Fixed for new installs with Beta 3
Please cleanup cache and user settings eventually
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

closed 2019-05-27 by trouch

k9carroll, 2019-05-28:
So I am hoping someone can help me answer a question.  I updated to Cura 4.0 and downloaded the Creawesome file.  I extracted the resources file and pasted it inside my Cura folder on my C dirve. When I start Cura the load screen shows the Creawesome logo but once the program actually starts there is no change.  There is no different in profile settings.  Also, I select my ender 3 and the screen does not show the actual machine in light gray in the background.  Does anyone have any idea of what the problem is?