trouch/CreawsomeMod #8: CR-X not supported?

TheMrCaveman, 2019-05-14:
CR-X not supported?

Are you willing to add Creality CR-X?

trouch, 2019-05-23:
To be honest, I haven't played with multi material yet. This is not a priority.
You can use CR-10S profile for single material.

lorak41, 2019-05-25:
Can't wait for github to allow sponsor anyone. I would definitely support your project. :) Hoping that you could add CR-X one day. Cheers

trouch, 2019-05-27:
@lorak41 ask Creality to send me a CR-X :-D

lorak41, 2019-05-28:
I upgraded my CR-10s to CR-X for $20. (driver + stepper motor + V-adapter)

trouch, 2019-05-28:
The issue is tinkering time, not the cost. I have a MM capable printer in project, but too many things ongoing :-)

AtlantaRusty, 2019-05-30:
Having the CR-X MM capability would be phenomenal in this mod, but using it as a single feed 10S for now is great.  Thank you for the mod.  Looking forward to getting better prints from my X, and hoping you can find the time in the future to add the X to the lineup.

Awesome job!

Beeez, 2019-09-19:
I second this!

trouch, 2020-03-21:
Please submit FR or PR to Ultimaker Cura repo

closed 2020-03-21 by trouch