trouch/CreawsomeMod #9: Is there a way to support Linux?

mhcerri, 2019-05-22:
Hi,

Awesome project! Do you think is there a way to support Linux as well? I believe Cura is shipped for Linux as a single AppImage file, so that probably make things harder.

Are you planning to release it as a regular Cura plugin maybe?

Thanks.

trouch, 2019-05-23:
I hadn't time to look into Linux version yet. ASAP :-)

fdraeger, 2019-05-23:
Hi there, 

I managed to install it on Linux (Linux Mint, but should work on Ubuntu and Debian derivates, too).
With Cura 4 installed, go to ~/.local/share/cura
Rename the folder 4.0 to 4.0_old
create a new folder 4.0 
change into this folder
extract all the files from the CreawsomeMod download into this folder.
you may remove the __MAXOSX folder

Starting Cura, you should now see a new startup splash window.

Cura seems to be very sensitive with extensions installed on top of CreawsomeMod. It complains about inconsistent config files and wants to reset to factory settings. If you do this, you loose the whole CreawsomeMod settings. - profile and all.

Cheers,

   Frank

mhcerri, 2019-05-24:
Thanks! I will give it a try

ltgrbrnz, 2019-05-24:
I tried Fdraeger's instructions.  Didn't work.  But.  I recreated my original 4.0 folder and simply copied the contents of the resource folder into the 4.0 folder, merging where requested, and it worked.  (At least so far. Printing test cube now.)  btw, I'm running Mint also.  Still copy the 4.0 directory to 4.0_old if you try this!  I'd hate to be responsible for any loss. Thanks Frank for the pointers.

probonopd, 2019-05-24:
Would you be interested in having an AppImage? I think I could help.

drewlio, 2019-05-25:
I just did this and it seemed to work. Slight modification to @fdraeger 's earlier instructions. 

Save any custom materials or profiles before doing this.

`rm -rf ~/.cache/cura/4.0`
`mv ~/.local/share/cura/4.0 ~/.local/share/cura/4.0-old`
`mkdir ~/.local/share/cura/4.0` 
`mv /path/to/unzipped/CreawsomeMod/resources/* ~/.local/share/cura/4.0`

Start the appimage, new splash screen, pick the CreawsomeMod Ender3...bobs your uncle.

This is Mint 19, Cura 4.0, Creawsomemod 0.2

probonopd, 2019-05-26:
A CreawsomeMod AppImage build is available at 

https://github.com/probonopd/CreawsomeMod.AppImage/releases/tag/continuous

__CAUTION:__ So far completely untested. I expect that some fine-tuning will still be necessary. I don't know if it can mess up your normal Cura's settings. If you fear that, please make a __backup__ of your `~/.local/share/cura/4.0` before you run this.

trouch, 2019-05-27:
Thanks for pointing that out !

closed 2019-05-27 by trouch

probonopd, 2019-05-27:
0.3.0 uploaded.

rdehuyss, 2019-05-31:
Will there be an CreawsomeMod Cura appimage based on Cura 4.1?

probonopd, 2019-06-01:
Sure, if I get feedback from users whether it works as intended.

ltgrbrnz, 2019-06-01:
<sarcasm> Well, you'll get feedback if it doesn't work!

ltgrbrnz, 2019-06-01:
Hmm that had a Sarcasm tag that actually disappeared. It was meant to let you know that somebody is listening.

rdehuyss, 2019-06-01:
Working for me! Still need to learn a lot about Cura as I'm an absolute
beginner. But I did finish my first successful build.

deepfriedcode, 2019-06-01:
Works well for me, thanks @probonopd.

probonopd, 2019-06-01:
New builds have been added to https://github.com/probonopd/CreawsomeMod.AppImage/releases, including 4.1.

mhcerri, 2019-06-06:
Awesome work, @probonopd ! Thank you!

gleonard3, 2019-06-12:
@probonopd, I use this on my Chromebook.  Works great.  Thanks for doing that.  I hope you stay on top of it.