trouch/CreawsomeMod #13: Broken settings - Had to nuke all my machines & settings.

tlhintoq, 2019-05-23:
After trying it and realizing I can't use my own nozzle sizes (only the preset sizes) I wanted to go back to my existing setup.

I returned my `resources` folder to the app directory and restarted.

Now my machine definitions are corrupt and I had to reset my entire installation back to factory defaults.  Not bloody happy.  Lost all my machines, materials, settings, preferences, add-ins.

Thankfully some of that gets brought back when you import a .3mf project.  But not all.

I think you need to provide much better technique for backing up ALL of someone's installation and how to restore it all back before asking someone to try your stuff.

[image]
[image]
[image]

XBrav, 2019-05-24:
It looks like it's trying to find the Creality CR10S5 files that were situated in the resources folder.

Have you tried reinstalling the resources folder over a clean install of 4.0? It should bring them back.

tlhintoq, 2019-05-24:
> Have you tried reinstalling the resources folder over a clean install of 4.0?
I haven't.  Wasn't interested in uninstalling/reinstalling at 2100hrs.  i just wanted to get something sliced and printing overnight.  So I let it reset the entire installation - which would have happened anyway on a re-install.
Then I opened a previously saved .3mf solution file which imported the settings and machine I wanted.  Then I sliced the new object.

torsoreaper, 2019-05-24:
I also had this problem.  I wanted to revert back to stock, closed cura, restored the resources folder, and when I opened Cura it noted the corruption and asked me to reset everything.  Luckily I only had about 4 or 5 custom profiles and only 1 I used often and I knew how to rebuild it so I just let it factory reset itself.  I also lost all plugins by doing this but whatever.

trouch, 2019-05-25:
As mentioned in the doc, there are no guarantee and you can break your setup. Really sorry for inconvenience. You really have to use Cura Backup feature first !
I'm closing this one since it duplicates several other issues.

closed 2019-05-25 by trouch