trouch/CreawsomeMod #15: Cura 4.0 crashes with the awesome resource

ieelan, 2019-05-24:
I replaced (added the new) resource folder and restarted cura 4.0 but it crashes.  Any suggestions to correct.  When I return the old (original) resource folder, cura 4.0 starts up.

jessearmstrong, 2019-05-24:
I have the same issue

jessearmstrong, 2019-05-24:
Here's the crash data that I get when trying to start with mod installed.

`Process:               cura [2853]
Path:                  /Applications/Ultimaker Cura.app/Contents/MacOS/cura
Identifier:            nl.ultimaker.cura
Version:               4.0 (4.0.0)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           cura [2853]
User ID:               501

Date/Time:             2019-05-24 08:27:39.797 -0500
OS Version:            Mac OS X 10.14.5 (18F132)
Report Version:        12
Anonymous UUID:        A66EF818-F292-F884-0126-10689FB00D77

Time Awake Since Boot: 62000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
dyld2 mode
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff76b622c6 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff76c1dbf1 pthread_kill + 284
2   libsystem_c.dylib             	0x00007fff76acc6a6 abort + 127
3   nl.ultimaker.cura             	0x000000010b88359e Py_FatalError + 462
4   nl.ultimaker.cura             	0x000000010b883378 _Py_InitializeEx_Private + 1336
5   nl.ultimaker.cura             	0x000000010b78aed7 main + 791
6   libdyld.dylib                 	0x00007fff76a273d5 start + 1

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x0000000000000000  rbx: 0x000000011b3b75c0  rcx: 0x00007ffee44754e8  rdx: 0x0000000000000000
  rdi: 0x0000000000000307  rsi: 0x0000000000000006  rbp: 0x00007ffee4475520  rsp: 0x00007ffee44754e8
   r8: 0x000000010b7afe10   r9: 0x00007fa894403310  r10: 0x0000000000000000  r11: 0x0000000000000206
  r12: 0x0000000000000307  r13: 0x000000010b9b2be8  r14: 0x0000000000000006  r15: 0x000000000000002d
  rip: 0x00007fff76b622c6  rfl: 0x0000000000000206  cr2: 0x00007fffad38d188
  
Logical CPU:     0
Error Code:      0x02000148
Trap Number:     133

Binary Images:
       0x10b78a000 -        0x10b931ff7 +nl.ultimaker.cura (4.0 - 4.0.0) <4E5DFBFF-A1A4-3774-837B-716108BF2954> /Applications/Ultimaker Cura.app/Contents/MacOS/cura
       0x11b315000 -        0x11b37f6ef  dyld (655.1.1) <CE635DB2-D47E-3C05-A0A3-6BD982E7E750> /usr/lib/dyld
    0x7fff4aac2000 -     0x7fff4af06fff  com.apple.CoreFoundation (6.9 - 1575.12) <64C38BF2-1869-33C8-B1B6-DCA9C824C0E4> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff735cf000 -     0x7fff735d0ffb  libDiagnosticMessagesClient.dylib (107) <A14D0819-0970-34CD-8680-80E4D7FE8C2C> /usr/lib/libDiagnosticMessagesClient.dylib
    0x7fff73a0e000 -     0x7fff73a0fffb  libSystem.B.dylib (1252.250.1) <1A13E822-B59C-3A36-A2E4-9968149714F9> /usr/lib/libSystem.B.dylib
    0x7fff73c51000 -     0x7fff73ca4ff7  libc++.1.dylib (400.9.4) <9A60A190-6C34-339F-BB3D-AACE942009A4> /usr/lib/libc++.1.dylib
    0x7fff73ca5000 -     0x7fff73cbaff7  libc++abi.dylib (400.17) <38C09CED-9090-3719-90F3-04A2749F5428> /usr/lib/libc++abi.dylib
    0x7fff74701000 -     0x7fff74962fff  libicucore.A.dylib (62135.0.1) <2B292561-798A-381C-AA41-90EE6549C7CF> /usr/lib/libicucore.A.dylib
    0x7fff75245000 -     0x7fff759cafdf  libobjc.A.dylib (756.2) <7C312627-43CB-3234-9324-4DEA92D59F50> /usr/lib/libobjc.A.dylib
    0x7fff760b4000 -     0x7fff760c6ff7  libz.1.dylib (70.200.4) <B048FC1F-058F-3A08-A1FE-81D5308CB3E6> /usr/lib/libz.1.dylib
    0x7fff768aa000 -     0x7fff768aeff3  libcache.dylib (81) <1987D1E1-DB11-3291-B12A-EBD55848E02D> /usr/lib/system/libcache.dylib
    0x7fff768af000 -     0x7fff768b9ff3  libcommonCrypto.dylib (60118.250.2) <1765BB6E-6784-3653-B16B-CB839721DC9A> /usr/lib/system/libcommonCrypto.dylib
    0x7fff768ba000 -     0x7fff768c1ff7  libcompiler_rt.dylib (63.4) <5212BA7B-B7EA-37B4-AF6E-AC4F507EDFB8> /usr/lib/system/libcompiler_rt.dylib
    0x7fff768c2000 -     0x7fff768cbff7  libcopyfile.dylib (146.250.1) <98CD00CD-9B91-3B5C-A9DB-842638050FA8> /usr/lib/system/libcopyfile.dylib
    0x7fff768cc000 -     0x7fff76950fc3  libcorecrypto.dylib (602.260.2) <BED0F385-D8A2-334A-BD7B-0673B5DABBC5> /usr/lib/system/libcorecrypto.dylib
    0x7fff769d7000 -     0x7fff76a10ff7  libdispatch.dylib (1008.250.7) <26DF5B1E-A388-38EF-B73B-AF0E93FB02DB> /usr/lib/system/libdispatch.dylib
    0x7fff76a11000 -     0x7fff76a3dff7  libdyld.dylib (655.1.1) <002418CC-AD11-3D10-865B-015591D24E6C> /usr/lib/system/libdyld.dylib
    0x7fff76a3e000 -     0x7fff76a3effb  libkeymgr.dylib (30) <0D0F9CA2-8D5A-3273-8723-59987B5827F2> /usr/lib/system/libkeymgr.dylib
    0x7fff76a4c000 -     0x7fff76a4cff7  liblaunch.dylib (1336.261.2) <BD5CDB35-2BB4-349A-8D14-78E387563F89> /usr/lib/system/liblaunch.dylib
[... 79 lines cut ...]
PCI Card: pci12d8,400e, USB Open Host Controller, Thunderbolt@140,0,0
PCI Card: pci12d8,400e, USB Open Host Controller, Thunderbolt@140,0,1
PCI Card: pci12d8,400f, USB Enhanced Host Controller, Thunderbolt@140,0,2
USB Device: USB 3.0 Bus
USB Device: Expansion
USB Device: USB3.0 Hub
USB Device: USB Storage
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB2.0 Hub
USB Device: Magic Trackpad 2
USB Device: USB3.0 Card Reader
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Display)
USB Device: Apple Thunderbolt Display
USB Device: Display Audio
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Display)
USB Device: Apple Thunderbolt Display
USB Device: Display Audio
Thunderbolt Bus: iMac, Apple Inc., 41.1
Thunderbolt Device: Thunderbolt Display, Apple Inc., 1, 26.2
`

Davinci522, 2019-05-24:
There is a secondary resources folder (lowercase) inside the first Resources folder (uppercase) you see. You need to replace the secondary lowercase one, not the first uppercase one. I had the same issue till I compared what folders were in them.

jessearmstrong, 2019-05-24:
That did the trick! Thank you!

trouch, 2019-05-25:
thanks @Davinci522

closed 2019-05-25 by trouch