trouch/CreawsomeMod #17: retraction speed

floxflux, 2019-05-24:
i need to change my retraction speed to 70 mm/s but unable to. how to fix?

torsoreaper, 2019-05-24:
I think it is limited to the factory stock marlin feedrate as outlined in the firmware as DEFAULT_MAX_FEEDRATE (X, Y, Z, E).

chunter1, 2019-05-24:
> I think it is limited to the factory stock marlin feedrate as outlined in the firmware as DEFAULT_MAX_FEEDRATE (X, Y, Z, E).

You mean Cura fetches the settings of the Marlin firmware on the printer?
I don't think so - at least i tried to change it in the firmware but Cura still didn't let me set it higher than 49?

torsoreaper, 2019-05-24:
> You mean Cura fetches the settings of the Marlin firmware on the printer?
> I don't think so - at least i tried to change it in the firmware but Cura still didn't let me set it higher than 49?

No, I mean i think the author looked in the stock Marlin firmware and just said "ok this is a hard cap".

I have my firmware set with a max E rate of 60mm/s but the Creawesome still wants to limit me at 50mm/so so it definitely is NOT reading anything from the printer.

floxflux, 2019-05-25:
if I was to change the retraction speed on the g code to 70 mm/s  from 50 mm/s manually (3.6mm length), do i need to only edit :

G92 E0
G1 F3000 E-3.6

into

G92 E0
G1 F4200 E-3.6

or i do need to edit every G1 F3000?

trouch, 2019-05-25:
> > > I think it is limited to the factory stock marlin feedrate as outlined in the firmware as DEFAULT_MAX_FEEDRATE (X, Y, Z, E).
> > 
> > 
> > You mean Cura fetches the settings of the Marlin firmware on the printer?
> > I don't think so - at least i tried to change it in the firmware but Cura still didn't let me set it higher than 49?
> 
> No, I mean i think the author looked in the stock Marlin firmware and just said "ok this is a hard cap".
> 
> I have my firmware set with a max E rate of 60mm/s but the Creawesome still wants to limit me at 50mm/s so it definitely is NOT reading anything from the printer.

Right ! All Creality firmware are caped at 25mm/s making any retract speed above that value useless.
Anyway the behavior is fixed in upcoming beta 3. There will be a warning above 50mm/s and hard limit at 200mm/s.

Mazo, 2019-05-26:
If you want to override it before the new version is out go to **resources/definitions/crealitty_base.def.json** and change **machine_max_feedrate_e**

floxflux, 2019-05-27:
> If you want to override it before the new version is out go to **resources/definitions/crealitty_base.def.json** and change **machine_max_feedrate_e**

thanks sir!

closed 2019-05-27 by floxflux