trouch/CreawsomeMod #18: Cura 4.1

h00etn, 2019-05-24:
Hello,

Cura 4.1 doesn't start when I put the resources folder in place and rename the original one. This is the error traceback info:

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/4.1/build/inst/bin/cura_app.py", line 136, in <module>
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 771, in run
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 211, in initialize
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 235, in _initialize
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 202, in _getBuiltinWelcomePagePath
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 85, in getPath
FileNotFoundError: Could not find resource ('WelcomePages/WelcomeContent.qml',) in 129

Ne0tuX, 2019-05-25:
The solution is shown on the last line of your log: copy the folder "WelcomePages" from yor backup folder to "C:\Program Files\Ultimaker Cura 4.1\resources\qml". I just tested, works fine.

trouch, 2019-05-25:
The mod is just not compatible with Cura 4.1 yet.
I actually need a mod version per cura release !

closed 2019-05-25 by trouch

niemyjski, 2019-05-29:
does beta3 support Cura 4.1?

ptyork, 2019-05-29:
> 
> 
> does beta3 support Cura 4.1?

No.

Revolt64, 2019-05-29:
Why are you closing these issues before they are resolved?

I greatly appreciate your work, but you shouldn't be closing issues until they're resolved, as it's a good place to track for users so we know when the issue in question is resolved.

Thanks for all your great work on the mod!