trouch/CreawsomeMod #20: Prime line

adrianmihalko, 2019-05-25:
I saw some people asked about this elsewhere and I would like to suggest too to use a "classic" prime lines: two long line on the left side of the plate.

conicio, 2019-05-25:
Add these lines to your start.gcode:
G28 ; Home all axes
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

trouch, 2019-05-25:
I admit the default purge line is less agressive and going to use it instead with beta 3.

adrianmihalko, 2019-05-25:
For me the problem with current one is it too short and it moves the mess to the print area. I had no problem with long skirt lines. I even have a velcro on the left bottom side, it helps cleaning the nozzle a lot.

trouch, 2019-05-27:
fixed with beta 3 !

closed 2019-05-27 by trouch