trouch/CreawsomeMod #21: Slicing problem

melou61, 2019-05-25:
Hello, I have a problem when slicing with your mod. As you can see in the pictures attached, in the prepare mode is all good, but when you change to the preview mode I cannot see the small details. Can you tell me how to solve this issue? Thanks
[image]
[image]

kaskii, 2019-05-25:
Try to enable "Print Thin Walls".

louis-lau, 2019-05-25:
Just FYI, if the above setting works it just means the details are smaller than your nozzle diameter/line width. Nothing to do with this mod.

melou61, 2019-05-27:
It works for now, thanks.

closed 2019-05-27 by melou61