trouch/CreawsomeMod #22: Marketplace

melou61, 2019-05-25:
As seen in the attached picture, I cannot get access to the marketplace plugins.

[image]

MrKaos98, 2019-05-25:
That should be internet connection related problem I think since it works fine for me. All plugins are visible..

melou61, 2019-05-25:
My internet connexion is fine. I closed and reopen Cura a few times and still the same, I've even reinstalled and nothing, still the same. I'm using MacOS Mojave 10.14.3

b1naryVap0r, 2019-05-25:
uninstall the cura 4, and reinstall. does can you access the market place without doing the creawesome mod? if yes, then close cura and  move the resources folder from the creawesome mod back into the cura directory, reopen and see if the behavior still exists?  it works for me on windows 10.

eptak, 2019-05-25:
Marketplace working fine !

melou61, 2019-05-27:
Thanks, it is working now.

closed 2019-05-27 by melou61