trouch/CreawsomeMod #23: No CR10S profile.

PrestonoftheDead, 2019-05-25:
I just downloaded this mod from the correct location in the text, not the Clone button.  I got it all spun up, and went to load my printer's profile, but it isn't there.  Any ideas?

PrestonoftheDead, 2019-05-25:
I should add, that there is a CR10S Pro, just not the basic CR10S.

PhillipMerritt, 2019-05-25:
If you already imported a CR-10S profile in your Cura installation it won't show up as an option. I just reinstalled Cura to fix this myself, but there might be a way to fix it w/o reinstalling.

trouch, 2019-05-27:
Fixed for new installs with Beta 3
Please cleanup cache and user settings eventually
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

closed 2019-05-27 by trouch