trouch/CreawsomeMod #24: Mac OS X Not working

Tarnag, 2019-05-25:
I followed the instructions for Mac OS X and Cura will not start with the modified resources folder, it immediately crashes and I have to go back to the old resources folder for it to launch again.

jwilkins728, 2019-05-25:
Same for me. I just downloaded and installed. It crashes right after I click the Cura icon

almarcano, 2019-05-25:
Try pasting the resources folder inside resources folder /cura/resources/resources, It solves for me...

James328, 2019-05-25:
Also not working for me on Mac OS X after following instructions.

High Sierra 10.13.6
Cura 4.0.0

gillesmartinit, 2019-05-26:
I have the same issue. 
After renaming the old and copy/paste the new folder /Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/resources, Cura doesn't show up anymore
 I found two kinds of errors in the console logs

`LSExceptions shared instance invalidated for timeout.`
`nw_path_close_fd Failed to close guarded necp fd 19 [9: Bad file descriptor]`
`nw_path_close_fd Failed to close guarded necp fd 19 [9: Bad file descriptor]`
`nw_path_close_fd Failed to close guarded necp fd 19 [9: Bad file descriptor]`

Mojave 10.14.5
Cura 4.0.0

xhros, 2019-05-26:
There is a resources folder inside the resources folder. As almarcano mentioned as well, you have to paste it in there. The directions here doesn't mention this as I had the same issue.

gillesmartinit, 2019-05-26:
> There is a resources folder inside the resources folder. As almarcano mentioned as well, you have to paste it in there. The directions here doesn't mention this as I had the same issue.

I renamed
`/Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/resources` 
to 
`/Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/resources.backup` and paste the resources folder of the zip in `/Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/`

So I have 
`/Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/resources` (the one from the zip)
and 
`/Applications⁩/Ultimaker Cura⁩/Contents⁩/Resources⁩/resources.backup` (backup of the orginal one)

Not working

xhros, 2019-05-26:
Clearly I can't read...lol. You just said that is what you did. I have been reading a lot of different people having issues, not related to this where they had to completely uninstall and reinstall it and it worked as expected.

gillesmartinit, 2019-05-26:
It's working
I reinstalled Cura 4.0.0 (I was in beta version) and copied the folder in the app and everything is working now
Thanks for your suggestion

Tarnag, 2019-05-27:
I tried this, it will now open, had to reinstall to get the Ender 3 to show up as an available printer, the profiles and mod still does not seem to be working.

trouch, 2019-05-27:
Fixed for new installs with Beta 3
Please cleanup cache and user settings eventually
https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations

closed 2019-05-27 by trouch