trouch/CreawsomeMod #25: Ridiculous naming

trouch, 2019-05-25:
This is an assumed typo. Feel free to suggest new name here !

b1naryVap0r, 2019-05-25:
you want a new name instead of "CreawesomeMod"?

if yes, i'd lean more toward incorporating: cura , creality & fixing prints. this way the name almost describes what it does.

CURE mod, becuase it sounds like cura, and cre(ality) and it cures (fixes) print settings

rivendellnetworks, 2019-05-26:
Don't change a thing...it's got character. I like it.

drock1985, 2019-05-26:
I do like the Creawesome name. It provides an easy and fun name for the mod; and does live up to the name. 

But if you truly dislike it that much I do like the concept that b1nary touched on. The mod provides symmetry between Creality and Cura. CreaCura? CreataCure? CreawesomeCure?

bkp23, 2019-05-27:
> I do like the Creawesome name.

I do, too, but that's not the name. It's "Creawsome", not "Creawesome". Note the missing 'e'. Certainly a typo.

Dev01-D, 2019-05-28:
It was a typo but is being kept to remind people not to take things so seriously!

hessius, 2019-05-28:
The name should probably not include "Cura" or "Creality", at least not as their own words, as that might infringe on their respective trademarks.

Mixing the names might be a good idea, like Creara. (I do like Curality as well, but dunno if it might be too similar to cura)

rivendellnetworks, 2019-05-28:
Will upgrading to the newly released 4.1 affect Creawsome?

stelgenhof, 2019-05-29:
@rivendellnetworks I suspect it will as this mod changes the installation of Cura.

nallath, 2019-05-29:
Ultimaker has a trademark on "Ultimaker Cura" not Cura. 

The way the mod does it completely breaks 4.1 (which is also how I ended up here, as I was investigating the huge number of crash reports we were getting...)

stelgenhof, 2019-05-29:
@nallath I questioned this mod (in particular its size) in this issue #48 . Although the modifications included are definitely resulting in better prints (from own experience), I think the method how these changes are applied is not appropriate. 

As you said, modifying/replacing the internal files of Cura may break user's installation when upgrading or even may cause Cura to stop working.

nallath, 2019-05-29:
Yeah. That's why I opened https://github.com/trouch/CreawsomeMod/issues/51. The crashes caused by this mod flooded our system and (as you might imagine) caused a fair bit of panic and stress with us Cura developers (Also because the error that it causes makes no sense as it's un a completely unrelated part).

These profiles should just be merged upstream in Cura. That way we know what profiles we need to take into account when changing things (eg; when we change things, make version upgrade scripts, etc). It also means that our unit & integration tests can actually be run over them.

trouch, 2020-03-21:
Thanks again @nallath for having integrated this mod, and sorry again for the inconvenience it brought to you.
Also thank you for keeping the major changes upstream. I saw a few things has been tweaker upon request, but we can't fight or argue for every single preference. So good so far with 4.4 and 4.5 !

Keep safe @ Ultimaker !

closed 2020-03-21 by trouch