trouch/CreawsomeMod #27: How to install CreawsomeMod on Cura for Linux?

sakvaua, 2019-05-25:
I'm not deeply familiar with Linux packaging systems. Cura is supplied as an AppImage. I cannot edit it directly. But I can extract the contents of the image into a folder, replace resource folder and make AppImage with https://github.com/AppImage/AppImageKit
But after I create a modified AppImage file it won't run cura. It just does nothing. No error or anything.
Any ideas what else I can try to mod cura for linux?

Roxsen, 2019-05-26:
I don't know about the AppImage as I don't use them but I installed cura through the repos on arch. my method was this:

**Install cura**
`sudo pacman -S cura`

**Find where cura resources were kept**
`whereis cura`

**backup current resources folder**
`sudo mv /usr/share/cura/resources /usr/share/cura/resources_backup`

**extract zip and copy resources**
`unzip CreawsomeMod-0.2.zip -d CreawesomeMod-0.2`

`sudo cp CreawesomeMod-0.2/resources /usr/share/cura/resources`

obviously your mileage might vary.

probonopd, 2019-05-26:
If there is interest from this project we could probably make a readymade AppImage. Just let me know.

probonopd, 2019-05-26:
Please see https://github.com/trouch/CreawsomeMod/issues/9#issuecomment-496018790.

stelgenhof, 2019-05-29:
@Roxsen I am running Arch Linux as well, and did the same steps as you described. Those are the correct steps.

probonopd, 2019-05-29:
Did you test the AppImage? A CreawsomeMod AppImage build for testing is available at 

https://github.com/probonopd/CreawsomeMod.AppImage/releases/tag/continuous

Credits go to @TheAssassin for the bunch of the work.

[image]

Greg-KB2QQM, 2019-06-03:
The instructions just the other day May 28 2019 on youtube  < https://youtu.be/mOn-VNqg2ac>
was **Not to install the clone or download link** but to download the Beta3 file. Here is that file link.. which should be updated. 

<https://github.com/trouch/CreawsomeMod/releases/download/20190527-0.3.0/CreawsomeMod-Beta3.zip>

  So that's what I did here with Linux Manjaro.  It's working fantastic.  My prints are absolutely amazing.  

**Install cura**
`sudo pacman -S cura`

**Find where cura resources were kept**
`whereis cura`

**backup current resources folder**
`sudo mv /usr/share/cura/resources /usr/share/cura/resources_backup`

**extract zip and copy resources**
`unzip CreawsomeMod-Beta3.zip -d CreawsomeMod-Beta3`

`sudo cp CreawesomeMod-Beta3/resources /usr/share/cura/resources`
(this did not work...so I just did it manually.   It worked)

**Open up both file locations as ROOT.  Go Copy (as Thunar Root in the right click menu) in the unzipped Downloads Folder and open up the extracted unzipped CreawsomeMod folder and drag that "RESOURCES" file into /usr/share/cura file location.     You need to be in Root in both files... put the "RESOURCES" file...   right beside the file named "resources_backup" **Done.

Roxsen, 2019-06-03:
@stelgenhof glad my instructions helped.

probonopd, 2019-06-03:
The AppImage should remove the need for those manual steps.

Roxsen, 2019-06-03:
@probonopd you are right they would remove the need for manual steps but not everyone likes AppImages, I know I don't. but thats the beauty of open source and Linux, do things the way YOU like. :)

stelgenhof, 2019-06-03:
Merging this mods enhancements into Cura itself removes the need for any manual steps, forks, offbeat hacks :)

Roxsen, 2019-06-03:
@stelgenhof there is that! :+1:

jrfoell, 2019-06-03:
I put together some instructions on rebundling the Ultimaker Cura AppImage with Creawesome on Linux: https://www.foell.org/justin/cura-creawesome-mod-for-linux/

probonopd, 2019-06-03:
> I put together some instructions

Thanks @jrfoell but are you aware that I am already [doing](https://github.com/probonopd/CreawsomeMod.AppImage/blob/master/convert.sh) that? So the user can just download the readymade AppImage.

jrfoell, 2019-06-03:
I didn't realize... I'll update my post with a link to your work (and fork).

probonopd, 2019-06-03:
If this project here likes my work I can send a pull request to have it integrated into this repository here, I'd actually prefer that over having to maintain a fork of the fork ;-)

trouch, 2019-06-03:
Thanks everyone for your contributions to make it work on Linux.
I've linked that thread on the README for reference and pinned it as well.

Feel free to submit a PR I could try in an Ubuntu VM; but I'll be honest this is not a priority.
Given the popularity and discussion with Cura devs, we are looking to include my definitions by default that would streamline the process. No mod, no specific build...

probonopd, 2019-06-03:
Indeed, I think that would be preferred by everyone! :+1:

WolfgangAxel, 2019-06-06:
The official Cura AppImage keeps configuration files in `~/.local/share/cura`. Each version of Cura you've installed makes a folder in this directory titled after the version number. This version number is the `resources` folder. I've successfully installed CreawesomeMod by doing the following in the past 5 minutes:

1. Download the newest Cura AppImage (currently [Cura 4.1](https://download.ultimaker.com/cura/Ultimaker_Cura-4.1.0.AppImage))
2. Download the CreawesomeMod for this version (4.1 beta currently [here](https://github.com/trouch/CreawsomeMod/releases/download/20190530-0.3.2/CreawsomeMod-Beta3.2-Cura4.1.zip))
3. Extract the CreawesomeMod zip
4. Copy and rename `resources` to `~/.local/share/cura/4.1/`
5. Set the AppImage to executable and launch

Full disclaimer, I have yet to actually print anything with it, but the mod is working. [Here's some quick screenshots](https://imgur.com/a/l9qiu09).

FWIW I'm on Debian testing, but that should be irrelevant with AppImages.

probonopd, 2019-06-06:
@WolfgangAxel why did you not just use the CreawsomeMod AppImage? No need to copy nor rename anything...

WolfgangAxel, 2019-06-06:
@probonopd I didn't use the AppImage because I didn't feel it was necessary. Your AppImage is a valid and functional way to have the mod pre-installed, but I didn't see a reason why GNU/Linux needed its own unique method of installation. I knew `~/.local/cura` existed, and I was curious if that's where the `resources` folder could be found. Lo and behold, it was there, just named differently. This means that the CreawesomeMod is installable in a nearly-identical manner to the other two main OS's listed in the README.

Yes, with your AppImage there's no need to copy or rename anything, but with using the method I used there's no need to trust another entity to extract, modify, rebuild, and reupload an entire AppImage after each update this way, even if it is automated.

I don't mean to sound rude. Both methods are functional and viable. One of them has 0 user input, the other doesn't require a 3rd party to modify 1st party files and redistribute them. Both have clear advantages and disadvantages related to convenience and security. It comes down to user preference, much in the same way that there are people who prefer to download prebuilt binaries and people who prefer compiling from source.

If nothing else, **we now have multiple options for installation**, which was the whole reason I made the comment in the first place. I wasn't trying to undermine anything you had done, I just wanted to notify the developers that the AppImage Cura builds doesn't have to change the method of installing the mod compared to the Windows executable builds or the Mac app builds.

Roxsen, 2019-06-06:
@WolfgangAxel thanks for finding a manual method for Linux without needing root/Sudo :)

probonopd, 2019-06-06:
@Roxsen just for completeness, the CreawsomeMod AppImage also does not need root/sudo...

Roxsen, 2019-06-06:
@probonopd as it should be for AppImage but you still have to deal with a 3rd party like @WolfgangAxel pointed out.

Greg-KB2QQM, 2019-06-10:
Honestly I have run appimage on Linux Mint and had no problems and I was running Linux Manjaro which is completely foreign to me and I was just trying to make the darn thing work.   I downloaded Cura 4.1 which crashed on Windows 10 for me...but I also just found out my Hexcore Intel Cpu, one of the cores is causing program crashes.  So I might try 4.1 Cura.  I turned off the core.   Computer Hardware.  Never buy a refurbished system.  ugh.

probonopd, 2019-06-10:
> Honestly I have run appimage on Linux Mint and had no problems a

Glad to hear!

Greg-KB2QQM, 2019-06-10:
After Using Linux Mint for a couple years and switching to Manjaro and having to learn terminal commands in Manjaro it was like learning French in German 😆

trouch, 2020-03-21:
thanks everyone, I guess you can now rely on Cura upstream !

closed 2020-03-21 by trouch