trouch/CreawsomeMod #30: Store the flow rate in the material profile

zebons, 2019-05-26:
In the same meaning that temperature, retracts parameters, flow'd be managed in the material profile.
If possible add the flow parameter in the material's profile.

BoatrightTBC, 2019-05-27:
Couldn't agree more. I have some filaments which hate to be rushed, and even with my microswiss all metal hot end, just won't go as fast as others. I would love to be able to adjust flow on a material by material basis.

Ghostkeeper, 2019-05-28:
It's not used very often, but Cura's material profiles can store a value for flow by using the `cura` namespace like so:

    <cura:setting key="material_flow">40</cura:setting>

Here's an example of a material profile that uses this type of setting key: https://github.com/Ghostkeeper/3D4MakersFacilianC8/blob/master/3d4makers_facilan_c8.xml.fdm_material

rlogiacco, 2019-05-28:
@Ghostkeeper do you scavenge the existing materials or is there a list of those keys published somewhere? All my searches have failed so far...

stelgenhof, 2019-05-29:
@Ghostkeeper Cura keeps these settings (flow) in the general cura.cfg file. What is the rationale for that? To me material flow seems an attribute of the filament and would think it belongs to the material settings. The same thing is the case for material price. (Just curious :) )

Ghostkeeper, 2019-05-29:
> @Ghostkeeper do you scavenge the existing materials or is there a list of those keys published somewhere? All my searches have failed so far...

The list of keys supported in normal `<setting>` tags is here: https://github.com/Ultimaker/Cura/blob/f6627daa49ecde38df115a88c8bfc61c834518e8/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L1171

The list of settings supported in `<cura:setting>` tags (is actually all Cura settings) is here: https://github.com/Ultimaker/Cura/blob/master/resources/definitions/fdmprinter.def.json

> @Ghostkeeper Cura keeps these settings (flow) in the general cura.cfg file. What is the rationale for that? To me material flow seems an attribute of the filament and would think it belongs to the material settings. The same thing is the case for material price. (Just curious :) )

I actually think that material flow is more of an attribute of the feeder on your printer rather than of the filament, though in reality it's probably a combination. And of course you also need to take into account the quality profile, because with higher layer heights you're going to extrude faster and then you'll have more slip as well.

The point I'm trying to make is that if you look closely enough, almost every setting is influenced by multiple components in the setting stack. That's why Cura doesn't make any assumptions and doesn't restrict certain settings to certain points in the setting stack. Restricting it would only make it harder on the profile developers and harder on the code complexity.

stelgenhof, 2019-05-29:
@Ghostkeeper Thanks for the explanation! I also agree it is a combination: there are multiple parameters that can be used to manage the material flow. Just wondering because other slicers (e.g. Slic3r) have this defined as an attribute of the filament.

zebons, 2019-05-29:
@Ghostkeeper I tried your solution (Cura 4.0 & Cura 4.1) to store Cura namespace parameter in a material profile but it dis'n' seem to work :

trouch, 2019-05-29:
thanks a lot @Ghostkeeper for pointing those informations out !
Cura wiki lack a bit of info and I haven't dig into all the source code yet.

Ghostkeeper, 2019-05-31:
@zebons Sorry, I forgot to mention that it also needs the namespace definition at the top of the file in the `<fdmmaterial>` tag then:

    <fdmmaterial xmlns="http://www.ultimaker.com/material" xmlns:cura="http://www.ultimaker.com/cura" version="1.3">

zebons, 2019-05-31:
@Ghostkeeper doesn;t work

Ghostkeeper, 2019-05-31:
Like I said, the Flow parameter will not enter in the profile screen:

> However it still won't show up in the Cura interface. The list of settings there is hard-coded in Cura.

But it will change the setting when you load the material. The point of this is that you then don't need to override the setting in all of the quality profiles.

zebons, 2019-05-31:
Ah Ok I undertstood the concept but as we cann't see the value in the profile material screen, not easy to manage it if we can only change it by editing the profile material file. 
By the way, we can use the MaterialSettingsPlugin and then see the choosen params in the profile screen

trouch, 2020-03-21:
thanks again @Ghostkeeper for your support and keeping the mod essence upstream !
hope you're all safe at ultimaker !

closed 2020-03-21 by trouch