trouch/CreawsomeMod #33: Cr-10s-pro

batz88, 2019-05-26:
Installed creawsome on the ender 3 pro prints fantastic no changes needed. Tried same print on my cr 10s pro really bad stringing am I missing something. Any help would be fantastic

peejayw, 2019-05-26:
I installed on my Ender 3 and got bad stringing, not sure what settings to
try to improve it.

b1naryVap0r, 2019-05-26:
makers muse did a video about stringing  ( https://www.youtube.com/watch?v=_QRb54zVPfQ ) 

Cura settings for the Ender 3 provided by @ChazMeister2 - https://goo.gl/vnsQWx, - note that you cant import the profile because of the changes the creawesome mod makes to cura, it doesnt recognize the profile as valid for import.
--
im not having much stringing on my ender 3 w/ stock creawsome settings. 

you should post what material you are printing, temp, speed. these settings, retraction settings along with outer wall wipe distance and cooling, impact stringing. ive heard that coasting can also help but i havent used it.

trouch, 2019-05-27:
should be better with beta 3 increased retract speed.

closed 2019-05-27 by trouch