trouch/CreawsomeMod #34: Skirt is not the first thing to print

ToddMangan, 2019-05-27:
When I start my print the skirt is not the first item to be printed

trouch, 2019-05-27:
seems a bug in Cura 4... removed skirt when support are enabled with beta 3 of the mod

Firedrops, 2019-05-28:
that doesn't sound like an ideal solution... is there a way to print bed adhesion before supports?

trouch, 2019-05-28:
Haven't found that !

jrfoell, 2019-05-28:
I'm seeing the same issue, a small part of the print is being laid down before the skirt.

Gaz83, 2019-05-28:
Not sure if it helps but I am printing brim and brim always prints before model.

tvixen, 2019-07-22:
I have no problems skirt. It's set to 4 pr. default, and it work out of the box.
Did also try with brim....works too.

trouch, 2020-03-21:
I believe it has been fixed in Cura since then.

closed 2020-03-21 by trouch