trouch/CreawsomeMod #38: Mac not working

maxfarnell, 2019-05-27:
I did everything it said, I renamed the file "resources_old" and copy and pasted the new one in. Rebooted Cura and no change what so ever. What am I doing wrong???
[image]
[image]
[image]

eptak, 2019-05-27:
I've just updated another mac at home without any issue !

maxfarnell, 2019-05-27:
Am i missing something or doing something wrong?

trouch, 2019-05-28:
having crash log or screenshot would be helpful

safetyABJ, 2019-05-28:
I'm having the same issue, clean install of Cura with new Creawsome mod installed crashes on startup: Cura 4.1.0, OS X 10.12.6

Error Traceback

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/Users/ultimaker/build/4.1/build/inst/bin/cura_app.py", line 136, in <module>
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 771, in run
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 211, in initialize
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 235, in _initialize
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/UI/WelcomePagesModel.py", line 202, in _getBuiltinWelcomePagePath
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/UM/Resources.py", line 85, in getPath
FileNotFoundError: Could not find resource ('WelcomePages/WelcomeContent.qml',) in 129

Logs

Thread 0x0000700012695000 (most recent call first):
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/selectors.py", line 376 in select
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/subprocess.py", line 1715 in _communicate
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/subprocess.py", line 1072 in communicate
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/OSXRemovableDrivePlugin.py", line 17 in checkRemovableDrives
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 59 in _updateThread
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700012192000 (most recent call first):
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700011c8f000 (most recent call first):
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 297 in wait
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x000070001178c000 (most recent call first):
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700011289000 (most recent call first):
  File "/Applications/Ultimaker Cura.app/Contents/MacOS/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 76 in _updateThread
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x0000700010d86000 (most recent call first):
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/UM/Backend/Backend.py", line 150 in _storeStderrToLogThread
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 862 in run
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/Users/ultimaker/build/env/4.1/inst/lib/python3.5/threading.py", line 882 in _bootstrap

[... 58 lines cut ...]
[... cut ...]
Current thread 0x00007fff9e4473c0 (most recent call first):
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 306 in _logInfoWidget
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 156 in _createDialog
  File "/Users/ultimaker/build/4.1/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 77 in __init__
  File "/Users/ultimaker/build/4.1/build/inst/bin/cura_app.py", line 102 in exceptHook

Roiberko, 2019-05-28:
Fix is easy, just copy from the backupdir/qml the directory "WelcomePages" to new resources directory

trouch, 2019-05-28:
4.1 is not supported yet

closed 2019-05-28 by trouch