trouch/CreawsomeMod #41: Missing parts in Print

Michalczyk-T, 2019-05-28:
Hello:) 
Thanks for the mod! I installed it and printed a filament swatch on my ender 3 pro (cura 4.0). 
after the print was done I realized that the numbers were not completely printed.
I looked up the preview and it is the same there. I attach some pictures of the issue along with the *.stl. I don't know if this is a bug of the mod or if this is something that could be fixed in the settings but I can`t find the error. loading the default profile and slicing it doesn't show this problem. Some help would be really nice.
Thanks a lot.

Best Tommy

[image]
[image]
[image]
[Eryone_Schwarz_PLA.zip](https://github.com/trouch/CreawsomeMod/files/3228518/Eryone_Schwarz_PLA.zip)

rlogiacco, 2019-05-28:
This is a consequence of the increased `line width` parameter: any detail below `0.5 mm` is not going to be printed using this mod. This is one of the downsides, but not the only one.

Try reducing the parameter back to `0.4 mm` and see if the preview gets fixed.

Michalczyk-T, 2019-05-28:
Hey rlogiacco.
Thanks for the response. You are right! The preview shows the numbers correctly now.
What is the downside of reducing the line width? Horizontal layer adhesion?
When reducing the line width would lifting the material flow value compensate for it?

rlogiacco, 2019-05-28:
No, you shouldn't change the flow rate unless you want to compensate for under/over extrusion. The author increased the wall line parameter to improve final print quality at the expense of fine details. This whole mod is a new approach to the trade-off quality/speed/robustness. As an example, this mod by default swaps the order of print for walls and infill: this increases the exterior quality of the model at the expense of internal structural robustness and sacrifices also on overhangs, which will be more prone to sagging.

There is nothing inherently bad in the author choices, he is just favouring exterior quality over other aspects. If I have to make a critique I would say he is not advising the downsides of the mod.

To be fair toward the author I have copied many of his settings creating an _Aesthetics_ profile which I plan to use for figurines and such, but I'm not going to use this mod as a whole.

trouch, 2019-05-28:
With regards to the the details loss, please read the tuning guide carefully.
https://github.com/trouch/CreawsomeMod#line-width

closed 2019-05-28 by trouch

trouch, 2019-05-29:
@rlogiacco 
> As an example, this mod by default swaps the order of print for walls and infill: this increases the exterior quality of the model at the expense of internal structural robustness and sacrifices also on overhangs, which will be more prone to sagging.

I partially disagree with this statement.
1. You are confusing perimeters order and infill-shell sequence. Printing _external_ perimeters first reduces overhang capability. But the mod prints internal one first with 3 perimeters to ensure good overhang. However, in some cases, I agree the infill can leave blobs on an overhang making the sagging to appear, mostly when the flow isn't calibrated.

2. With regards to internal structure, if you think about infill-shell-infill-shell... you can fairly understand you always have an infill between two shells no matter of the order, unless you combine infill layers. The big difference is on the infill-shell link. I agree printing perimeters lastly on each layer will make a better link by shielding the infill. But we also need to closely look on the infill-shell overlap defined with `infill_overlap` settings, by default set to 10%, increased to 30% in the mod. Making very robust link. Unfortunately I don' have a microscope to check it closely.

At this point, getting a robust part is mostly about configuring proper infill density/pattern/direction as well as printing orientation and getting pretty one about calibrating your temperatures and flow.... hopefully :-)

Michalczyk-T, 2019-05-29:
Thank you, this is realy helpfull!
it realy is an Awesome Mod!