trouch/CreawsomeMod #45: Changelog?

drock1985, 2019-05-28:
Hi,

I see that the mod was updated to Beta 0.3 and I was wondering what the changes were between 0.2 and 0.3 versions?

Thanks,

RFC1795, 2019-05-28:
For one, I see I'm now purging from the left side rather than the front. I also got some message pop-up when I started the new profile, I dismissed. Something about my firewall complaining about some setting, I dismissed that too. First print I had to cancel, I might go back to the previous release after a few more tries.

1aTa, 2019-05-28:
Just click on releases.

https://github.com/trouch/CreawsomeMod/releases/tag/20190527-0.3.0

trouch, 2019-05-28:
> Just click on releases.
> 
> https://github.com/trouch/CreawsomeMod/releases/tag/20190527-0.3.0

closed 2019-05-28 by trouch