trouch/CreawsomeMod #47: Ender 3 Bed Size

stelgenhof, 2019-05-29:
This has been a discussion in the past in the community and there seems to be two camps :)

The Cura provided Ender-3 profile defines the bed size as 235x235 mm, however this package defines it as 220x220 mm. I am aware that 220 is the official size (as defined by Creality), however a lot of people in the community 'pushed' to have the size set to 235x235 as that is what in reality can be used.

Just wondering why 220x220 was chose in this package.

Cheers! Sacha

trouch, 2019-05-29:
Has been set to 235x235 in beta 2, people asked to use spec size.
Set to 220x220 and now people ask to get 235x235 😅 
But now you can adjust machine size for your prefer right after adding the machine.

closed 2019-05-29 by trouch

stelgenhof, 2019-05-29:
Haha, I experienced the same when creating the default Ender-3 profile. Thanks!