trouch/CreawsomeMod #51: Merge into Cura

nallath, 2019-05-29:
Is there a specific reason that these profiles cannot be merged directly into Cura? If these do get merged into Cura, no manual action is needed to get these profiles (and it would also mean that we don't get any startup crashes anymore)

trouch, 2019-05-29:
Pushing upstreaming is not planned for now since it breaks previous personal profiles usability. 
Also I'm pretty sure the "Creawsome" or any category different than "Other" isn't allowed.
Lastly, it is not clear when exiting definition have been provided by the community or Ulitmaker itself. I don't want to override someone else contribution and trash it.

nallath, 2019-05-29:
Ultimaker didn't provide any definition apart from the Ultimaker profiles. So anything in other is in that area. The popularity of this mod makes it pretty damn clear that your profiles are vastly better than anything that was contributed before. 
As far as personal profile usability: This shouldn't break at all. As long as the same ID's are kept everything should work just fine. 
So this essentially just leaves the category issue, which doesn't feel like a large enough "feature" to warrant all the issues the current setup is causing. 

This mod is causing **100-30 crashes per hour** right now (and caused quite a bit of panic with us Cura Developers, as we got scared that we had an exploding cura release going on). I'm really hoping we can get these profiles merged because this might cause the same (or similar)  issues again next release.

bkp23, 2019-05-29:
Trouch, I'd urge you to reconsider this issue. It's not often that you get to work directly with the developers of a program you're modding. In this case, they can help provide a long-term approach for better stability and sustainability.

b1naryVap0r, 2019-05-29:
Good on ultimaker for reaching out, would be cool to see everyone benefit the combined efforts, without trouch losing credit for his effort

nallath, 2019-05-29:
People not getting the credit that they deserve is the bane of every open source project. Even though 90-95% of the development of Cura is paid for by Ultimaker, doesn't mean it's not an OS project. I actually want more people to contribute to Cura because that would benefit everyone (Ultimaker included).

I'm actually happy that someone put so much time into optimizing profiles (and a little less happy that we found about it because of the whole "omg so much crashes, wtf is going on" panic we had this morning ;))

trouch, 2019-05-29:
Hi @nallath sorry it wasn't clear you are actually from Cura.
Please receive my all apologies for the crash reports increase.
I tried to warn as much as possible it would break upon upgrade, but it became popular with quick !

trouch, 2019-05-29:
I'll be publishing a new package compatible with Cura 4.1 later today, I've ran some successful test;
With regards to next Cura release, I can submit a PR upstream. 

At the beginning the plan was to make a full build, forked from Ultimaker but I running into issue with mojave. All my tests failed so far, and seems I'm not the only one.
https://github.com/Ultimaker/cura-build-environment/issues/51

trouch, 2019-05-29:
https://github.com/Ultimaker/Cura/pull/5809

fiveNinePlusR, 2019-05-29:
how would this work in the future? just have to issue prs to the upstream cura?

trouch, 2019-05-30:
> how would this work in the future? just have to issue prs to the upstream cura?

yeah that's the plan to get integrated within Cura out of the box.
the good news - no more issue upon update or dirty modification.
the bad news - depends on Cura release schedule which is out of control from 3rd party developer.

Assuming the PR get approved, Beta 3 is probably the last version of the mod within its current form. 
I would have wish more community testing, feedback, and additional iterations to adjust the settings before pushing upstream, but the adoption went out of control in few days.

I still believe a mod or Cura fork is nice to experiment new things in the community without involving Ultimaker.

Ghostkeeper, 2019-05-31:
> the bad news - depends on Cura release schedule which is out of control from 3rd party developer.

Allowing publishers to put printer profiles on Cura's Marketplace is on our road map, but won't be there for a few releases to come still. It'd work a bit like how the Marketplace currently provides material profiles. Then you could publish new versions on your own release cycle.

trouch, 2019-06-01:
> Allowing publishers to put printer profiles on Cura's Marketplace is on our road map, but won't be there for a few releases to come still. It'd work a bit like how the Marketplace currently provides material profiles. Then you could publish new versions on your own release cycle.

That would be perfect (not say _awesome_) ! I have some ideas I'd like to share with you and willing to help on that topic if you are interested.

Ghostkeeper, 2019-06-03:
Sure, but probably better to do that in the Cura repository though.

trouch, 2019-07-03:
Merged with https://github.com/Ultimaker/Cura/commit/e8c08b77e597332251e837fa89d4a9a9179e5338

closed 2019-07-03 by trouch