trouch/CreawsomeMod #54: Cura start gcode settings changes not being kept.

yfaykya, 2019-05-29:
Hi,
I add a G29 to start gcode but anytime I look at the generated gcode it is not there.

trouch, 2019-05-29:
cannot reproduce it !

b1naryVap0r, 2019-05-30:
I have g29 on my install for bltouch and works without issue after multiple prints.

Maybe there was something you had to do to cura initially to configure auto bed leveling and it was removed during the mod install?

Try to uninstall and then reinstall cura with no mod to confirm it functions then install the mod to confirm you still have the problem.

yfaykya, 2019-05-30:
Only thing I did was edit the start gcode in printer settings. I will have
another dig at it today to see what I may have missed.

yfaykya, 2019-05-30:
Working now. I might have had multiple curas open or something. Closing. (and thanks!)

closed 2019-05-30 by yfaykya