trouch/CreawsomeMod #55: Ender 3 Stringing and Ghosting

Marstaw, 2019-05-29:
Hi,
I just printed a calibration cube with the settings and got ghosting on y axis. 
Then I printed a benchy and there was alot of stringing issues.
Am I doing something wrong lol ?

ArturNadolski, 2019-05-29:
Same here. In my case changing retraction to 7 solve the issue.
On machine with bowden extruder You can try to enable coasting (experimental).

trouch, 2019-06-02:
1. Make sure to calibrate your extruder and flow first
2. Calibrate the temperature
3. Adjust your retract, 5mm at 45mm/s should work if 1 and 2 are properly done.
You can follow the calibration guide  linked in the doc.

With regard to ghosting, the mod settings are really low to avoid it. 
Maybe your table isn't stiff enough.

closed 2019-06-02 by trouch