trouch/CreawsomeMod #56: CreawsomeMod Beta 3 (Cura 4.1 Only) configuration error

1aTa, 2019-05-29:
Cura 4.0 removed along with all config files and clean install of Cura 4.1 + CreawesomeMod.

When duplicating Generic PLA I see the following error:

Your config appears to be corrupt. Something wrong with the following profiles: generic_pla_175_hms434_0.4mm_thermoplastic_extruder

trouch, 2019-05-29:
What OS ? Windows ? Haven't double checked on that one this one.
Just tested on macOS with another computer without issue.

1aTa, 2019-05-29:
Windows 10 1903.

Guilouz, 2019-05-29:
Same here on macOS.
Error when duplicating duplicate generic material but still duplicated.
However, it is impossible to delete the duplicate material.

[image]

[image]

1aTa, 2019-05-30:
It's not impossible, you just have to make a different material active.

You can't delete a material which is active and in use.

trouch, 2019-05-30:
I've been able to reproduce the error. Not sure about the root cause yet.

trouch, 2019-05-30:
Should be fixed with Beta 3.2.

1aTa, 2019-05-30:
Great, thank you.

FYI typo CreawsomeM**i**d-Beta3.2-Cura4.1.zip

trouch, 2019-05-30:
> FYI typo CreawsomeM**i**d-Beta3.2-Cura4.1.zip
😬 fixed 🙈

closed 2019-05-30 by trouch