trouch/CreawsomeMod #61: Gaps at start of each print line

gerhick, 2019-05-30:
Hi all, I upgraded to creawesome yesterday and it looks great, but I'm getting a problem where the start of each line has a slight delay, maybe 1 or 2mm and its consistent - see attached images. 

I thought it was related to retraction but I've disabled retraction and its still happening - albeit less so...

Note in the images attached the part orientation is the same, the break happens top right in each case...

Any suggestions?

[image]
[image]

malikobaid, 2019-05-31:
Having the same issue after update

trouch, 2019-06-06:
this is typical to bed adhesion or levelling issue.

closed 2019-06-06 by trouch