trouch/CreawsomeMod #63: 4.1 CreawsomeMod does not open (Mac)

ren9216, 2019-05-30:
Hi! I am having the same issue as SimplySomething. After loading it takes me to the reset nad backup configuration, and it wont open. I tried reseting cura already updated and then reinstalled the mod, but does not work.
I am new at this so detailed explanation on how to do it would be amazing!
Thanks!

ToneStrife, 2019-06-02:
Same to me. Cura 4.1

BarbaRufa, 2019-06-02:
Exact same here.  Downloaded a fresh 4.1 copy from Ultimaker's web site. Downloaded the 4.1 mod and unfortunately cannot open.

ToneStrife, 2019-06-02:
Found the answer. Instructions are wrong. The folder to copy is resources, inside of resources.
Cura.app/Contents/Resources/Resources

BarbaRufa, 2019-06-03:
ToneStrife. you are a genius. Thanks.

trouch, 2019-06-03:
Doc is as clear as possible with a screenshot and arrows. 
[image]

Look ... Contents.... Resources (capital R) then ressource (lower R)...

closed 2019-06-03 by trouch