trouch/CreawsomeMod #64: prints are 0.2 layer height no mater what profile i  select

davelandrygmail, 2019-05-31:
no matter what profile i select my layer heights are always 0.2

davelandrygmail, 2019-05-31:
Performed some more investigations with a fresh install on a new PC.

The issue is Dynamic Quality.  Every other profile has Layer Height = Initial Layer Height except for Dynamic where Layer Height = 0.16 and Initial Layer Height = 2.0 

So this might be a bug with Cura.

trouch, 2019-06-04:
Could you provide an STL to replicate the issue ?

davelandrygmail, 2019-06-05:
good morning please see attached files

[image]

[CCR10_CHEPCalibrationCube.gcode.txt](https://github.com/trouch/CreawsomeMod/files/3257114/CCR10_CHEPCalibrationCube.gcode.txt)

trouch, 2019-06-06:
this model is mostly vertical, but stills show some layer height change on the preview.
Please use another STL with more slopes to see more changes.
[image]

closed 2019-06-06 by trouch