trouch/CreawsomeMod #66: GCode keywords not filled in

NullPointerLabs, 2019-06-01:
To facilitate auto bed leveling on my printer I have modified the start-up GCode to heat up the bed, perform ABL and then heat up the nozzle (to prevent oozing while performing ABL). I have added the `{material_bed_temperature}` and `{material_print_temperature}` keywords to heat the bed and the nozzle when needed. Looking at the generated GCode, Cura detects the keywords and omits the default bed/nozzle heating GCode it would normally "inject" at the start of the file. However, the actual keywords are never replaced and the generated GCode simply contains `M190 S{material_bed_temperature}` and `M109 S{material_print_temperature}` where the temperature should be.

trouch, 2019-06-02:
For what printer ?

NullPointerLabs, 2019-06-02:
This is happening for my Creality Ender 3 Pro, which I selected the CreawsomeMod profile for.

maludwig, 2019-07-08:
I'm having no troubles, I pull the same trick to avoid waiting, and I add an extra home to let myself know "yo the bed is hot now, the print is about to begin!" I chose 165C as the pre-heat extruder temp because that's where I got no oozing, and it's only a 20s wait to get to 190C, which I'm happy with.

It would be cool to have these as defaults, save everyone some time.

PRINTER SETTINGS -> Start G-Code:
```

M140 S{material_bed_temperature} ; start heating the bed to what is set in Cura and do not wait
M104 S165 ; Set extruder to 165C and do not wait
M190 S{material_bed_temperature} ; Wait for bed to heat to what is set in Cura
G28 ; Home before heating extruder
M109 S{material_print_temperature} ; Heat the extruder to what is set in Cura and wait

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
...
```

OUTPUT:
```

M140 S70.0 ; start heating the bed to what is set in Cura and do not wait
M104 S165 ; Set extruder to 165C and do not wait
M190 S70.0 ; Wait for bed to heat to what is set in Cura
G28 ; Home before heating extruder
M109 S190.0 ; Heat the extruder to what is set in Cura and wait

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
...
```

maludwig, 2019-07-08:
I'm also running the Ender 3 pro, btw. OSX Mojave, if that matters.

closed 2020-03-21 by trouch