trouch/CreawsomeMod #67: Increase overhang angle supports on ender 3

BenNottelling, 2019-06-01:
I'm talking about this setting
[image]

I did a angle test print with this profile and got some really nice results (from 30 - 85 degree angles)
So I propose the default setting on this be increased to maybe 60 or 70 degrees
[image]

I printed with hatchbox PLA at 200c on the hot end and 50c on the bed.

trouch, 2019-06-02:
please give more info 
- layer fan type - stock or modded
- layer fan duct - stock or modded
- fan speed
- print orientation

BenNottelling, 2019-06-02:
My bad
Fan type is and duct are both stock
Speed I believe is the default for this profile
Print orientation is the same as it's seen in the pictures (otherwise this test would be pretty useless) 
Can I run another print to double check the print speed if you want

trouch, 2019-06-02:
Thanks, but there are two pictures with different orientation on the Z axis. was it aligned with Y axis ?

BenNottelling, 2019-06-02:
Ohhhh that's what you meant, yeah I'll have to start the print again so I can see what direction it's printing

trouch, 2019-06-02:
airflow direction has big impact on the overhang !

BenNottelling, 2019-06-02:
Here I did a reprint with it facing both directions, I only printed from 50-85 degrees so it would save time, but I kept the same settings as last time, does that help?
[image]
[image]
[image]

trouch, 2019-06-06:
honestly, increasing to 75° doesn't sound a good idea unless super efficient cooling.

closed 2019-06-06 by trouch