trouch/CreawsomeMod #69: Issues found with settings so far on Creawesome that need tweeking

PriestWithKnives, 2019-06-02:
125% line width is wrong, it's 110%-120% line width, generally .44 is a reasonable figure for this. S3D for example has their Auto nozzle configuration at .48 or 120% which is the high side of this. This is partially why you have people complaining about gaps.

Zhop, turn it off unless you have issues with supports and that's from overextruding then. If you need to have it set to on at least run it at .12 to reduce some stringing

Infill/Outside pattern, Set to Infill first then Outside to give strength.

Infill pattern, Grid is the worst infill pattern, try changing it to Octet for strength and support in Z, added benefit that 10% octet is = 20% grid, so you could get away with less percent there.

Combing, Turn on, Within Infill, Max distance 30 mm/s, this will cure the rest of the stringing 

Infill Wipe Distance, set it .1

Besides that you added decent sanity checks to speed, jerk, and are using the thread pitch and stepper rotation magic numbers in .04 intervals. I will mention after seeing many prints the lowest or .04-.08 should be .05-.08, it can't do .04 reasonably most time.

Source, I make a lot of profiles for people in the community and most of the above we've tested and found them to work. I believe you can easily tweak them to get better quality without sacrificing much time wise.

Example prints with above changes in effect below that are default settings for me at this point. Give them a try sometime:

https://imgur.com/a/h6751tn

https://imgur.com/a/2w0qLKP

https://cdn.discordapp.com/attachments/358695610804600833/478393404489859092/IMG_20180812_224305.jpg

https://i.redd.it/4htinqov08w21.jpg

trouch, 2019-06-02:
Thanks @PriestWithKnives for this comprehensive report !

Before diving into details, please let me give some background. CreawsomeMod has been done on my personal preferences and experimentations to achieve best looking and accuracy no matter of the time spent in printing. Also I have no particular issue on my printed parts strength neither stringing. This is a personal project I shared to help people with a good (but not perfect) starting point. Very few people was using it so far without any issue. The big spike and hype wasn't expected, which brought lot of issues and bad comments by lack of proper tuning for most of cases.

It hasn't been made to fulfil all users needs and expectation, neither being the best settings. Also instead of tweaking all Cura settings, it just needs basic and mandatory machine/material calibration, but it has to be done properly.

Having said that, and given Cura dev team request to merge my definition, I now need to accept users requests and common consensus despite of my own preferences. I can still make my private fork 😅 

> 125% line width is wrong, it's 110%-120% line width, generally .44 is a reasonable figure for this. S3D for example has their Auto nozzle configuration at .48 or 120% which is the high side of this. This is partially why you have people complaining about gaps.

To what I found after lot of research, tests, and discuss here and there, the actual limitation is how you can consistently extrude, which is a volumetric speed given by the hotend/nozzle, for exemple 10-15 mm3 given E3D spec. We can go up 200% line width. I've introduced 125% for convenience right before first beta as I started to design my own models. I concur it makes troubles for people since lot of models found on Internet are designed for a 0.4ish width. I'm sure we can find a tradeoff here - 0.45 should be a good balance as per my recent testing. I may need to set width per nozzle variant instead of using a 125% rule.

> Zhop, turn it off unless you have issues with supports and that's from overextruding then. If you need to have it set to on at least run it at .12 to reduce some stringing

I understand people don't like Z-hop as it makes the print to take longer. But most of people are also disabling Z-hop to wipe the nozzle and avoid stringing instead of fixing root cause, basically material calibration. Also Z-hop requires proper mechanic adjustments, it doesn't like Z backlash and binding.

> Combing, Turn on, Within Infill, Max distance 30 mm/s, this will cure the rest of the stringing

Naaaa stringing is not a reason to use Combing to me, but you may get it with my previous answer.
I found there are only several combination of Combing/Z-hop/Retract that can be used.
For exemple, disabling combing requires to enable Z-hop. And if you want to disable Z-hop, you need to enable combing. Z-hop with proper retract is the best way to achieve perfect complex first layer for technical parts. That way you can have lot of very small holes in your parts. And once you start using Z-hop, you find it only hurts the print time. Anyway, since I'll probably disable Z-hop, I'll have to enable combing in such case. Actually a simple `Combing = !Z-Hop` rule could work.

> Infill/Outside pattern, Set to Infill first then Outside to give strength.

This is a statement I don't understand to be honest. `... skin - infill - skin - infill - skin ...` there are always an infill between two skins no mater of the order within a layer. Printing infill first shouldn't make it stronger, or I miss something, maybe when Z-hop is off and combing is on. What I'm sure - printing skin first avoids to see infill from the outside. 

With regards to infill-shell link, the mod increases the overlap from 10% to 30%, which should make strong links.

> Infill pattern, Grid is the worst infill pattern, try changing it to Octet for strength and support in Z, added benefit that 10% octet is = 20% grid, so you could get away with less percent there.

Infill switches from `Grid` to `Line` above 25%, it uses default Cura behaviour here. Having this settings make infill moves linear from walls to walls, and reduces overall vibrations. Tradeoff here. I don't think it's a critical choice, and each people have their own preference. Best choice would be honeycomb for strength, but it's not available into Cura. Gyroid is the worst to me. Maybe Cubic could a good candidate if you like X/Y pattern variation.
 
> Infill Wipe Distance, set it .1

What for ? I'm surprised you aren't talking about `Coasting` and `External wall wipe distance` instead to reduce the Z-seam.

In general, the best fix for all the issues with strings and blobs are Linear Advance and retraction both finely tuned. But this is not a new comer thing, and disabling z-hop is a quicker workaround instead of fixing root cause. Basically the bowden giving too much elasticity in the nozzle pressure with retract and printing move speed change.

> Besides that you added decent sanity checks to speed, jerk, and are using the thread pitch and stepper rotation magic numbers in .04 intervals. I will mention after seeing many prints the lowest or .04-.08 should be .05-.08, it can't do .04 reasonably most time.

Printing all walls at the same speed is a sacrifice to ensure stability and reduce vibrations that makes inconsistency. A small defect on an inner wall will be visible on the outer one. 
Also I prefer to talk about `Resolution` which has a unit (mm/step) and `Common Multiplier`, those are mathematical and physical words.

I'm going to do some tests with your suggestion, and keep you posted !

PriestWithKnives, 2019-06-02:
Hi,

Thanks for in depth reply.

These are mostly changes I suggested to people on the discords/reddit on how to fix issues they've seen in printing since upgrading to the mod. Tend to find the after posts are more positive about stringing issues so they might be good to test them as you said and see how they do comparatively which is why I brought them up. 

"Infill/Outside pattern, Set to Infill first then Outside to give strength."

In regards to that what I was talking about the setting Outer before inner walls checkbox .

Infill type, octet has been the best strength vs time I've found, also line infill at 25% higher is increasing your times as it goes up. Setting to to say 12% octet should make it comparable and save some time on printing overall. Probably not by a huge margin but might be worth a look.

As far as Z seam goes user defined is best and what I use with coasting to help reduce the Z scar to mostly non noticeable amounts even on curves, example

https://imgur.com/a/uNUvGgI

Fairly easy to dial in so didn't think it needed to be mentioned but if you have coasting off by default on could be a good idea. A minor coast doesn't hurt most prints and tends to help negate the z seam to just a minor scar.

Also resolution is a fine term but it's layer height in Cura so that's what I use as people know what it means. Yes thinner layers increase resolution but layer height is in the program so that's how I reference it. Having to explain increase the resolution to .12mm would probably leave half the people I help out scratching their head for a minute.

trouch, 2019-06-03:
> "Infill/Outside pattern, Set to Infill first then Outside to give strength."
> 
> In regards to that what I was talking about the setting Outer before inner walls checkbox. I understand what you're saying about the route from outer to inner should make a better surface but on some geometry it doesn't. Bridging is the biggest problem there. I've run it both ways but come back to inner before outer and haven't had issues since.

You are getting confused here !
Infill -> Infill Before Walls : That's what I'm talking about and I thought you too.
Shell -> Outer Before Inner Walls : I'm not going to activate that ! Never.
 
> Infill type, octet has been the best strength vs time I've found, also line infill at 25% higher is increasing your times as it goes up. Setting to to say 12% octet should make it comparable and save some time on printing overall. Probably not by a huge margin but might be worth a look.

As per my last time tests and evaluation, there are no major difference in print duration when using Grid or Octet or Cubic. Also enabling Z-hop only increase print-duration by ~5% on a 20h print. That is way less than I expected.

> Also resolution is a fine term but it's layer height in Cura so that's what I use as people know what it means. Yes thinner layers increase resolution but layer height is in the program so that's how I reference it. Having to explain increase the resolution to .12mm would probably leave half the people I help out scratching their head for a minute.
> 
> I think you've done some good work here. Yes calibration and tuning would be great as there is no silver bullet and it might be too much to ask for a way to add in estep calibration how to guide, where they run a 100mm test and a hollow cube after but that would be a cherry on the mod cake. Or just link say Matt's extrusion calibration while they set up the mod, then they'll have an on point esteps to work off and the whole z hop issue could be sidestepped. Again might be too much to add in but a few lines in the readme might make the profile shine on people who've never run one before.

Honestly, we should educate people in doing right things, using rights terms. We can ease users life and educate them. Those are not incompatible.

`Magic Number` and Z-hop workaround for strings are trendy because of non scientist Youtubers who got popular. If these guy would have learned how to properly calibrate your extruder (cold filament) and flow instead, everyone would benefit from it.

PriestWithKnives, 2019-06-03:
> "Shell -> Outer Before Inner Walls : I'm not going to activate that ! 

I don't want you to? That's what I said in my post? Did you read? I saw it checked on in your profile. I was bringing it to your attention...

> "As per my last time tests and evaluation, there are no major difference in print duration when using Grid or Octet or Cubic. Also enabling Z-hop only increase print-duration by ~5% on a 20h print. That is way less than I expected."

There are differences from grid and octet at the same percentage in printing, hence why I was bringing up 10% octet vs 20% grid infill strength.... Please give some figures for testing times if that's the case so we can "test" and get an get an accurate representation on your figures and see what's wrong there.

Secondly, I have not mentioned Z hop increasing time in layer? You have. Twice. It isn't my point, maybe you're confused? The point I made was stringing. Which I've mentioned now, twice.

Example picture from the comment below mine in this github to show you what strings look like

https://user-images.githubusercontent.com/17304943/58766257-36c0af80-8531-11e9-8503-51ebfcad92d5.jpg

trouch, 2019-06-03:
> "Shell -> Outer Before Inner Walls : I'm not going to activate that !

Just wanted to make sure we are on the page here 😃 

With regards to the Z-hop comment and print duration, I did at first myself, and it's a common complain, not from you. So I wanted to check that now the estimate are correct. Here are the numbers I've summarized. Model used is the famous rude gnome with a middle finger scaled à 200% (30cm height)

[image]

PriestWithKnives, 2019-06-03:
I'm glad you're answering your own questions here in regards to a gnome with a middle finger as your testing model on if z hop causes a great increase.

However, I was bringing up the point of stringing, as the example pictures posted shown above and the issue comment before mine also showed. 

The response to most from people asking for help has been disable z hop in order to get rid of this, so far it's been working. Might be something to try instead of measuring zhop time with 1 finger in the air.

trouch, 2019-06-03:
As I said, I'm open to adapt to user expectation, but I need to test and do my homework, even on my own statement. I already have a private branch somewhere else with your suggestions to fix stringing.

PriestWithKnives, 2019-06-03:
Glad to hear you are willing to listen! Here's some of the community feedback from today 

https://www.reddit.com/r/ender3/comments/bw00t8/problems_with_creawesome_mod_settings_and_ways_to?sort=confidence

PriestWithKnives, 2019-06-03:
Sounds Good, Luck with the mod

trouch, 2019-06-05:
Quick update...
- Line width : difficult to find a best tradeoff. Seems 105% might be better to print finest details, but no matter of the value, some walls might be replaced with infill depending on the model tolerances. 

- Stringing : I'm really not able to reproduce any major issue with the 10S Pro. But I am able to reproduce it with Ender-3. Having an enclosure and printing with door closed increase stringing a lot ! Since it's easier to put the Ender-3 into an enclosure, I guess that's why there are more complains with strings compared to 10/S/Pro.

- Z-hop/Combing : so far so good, I was expecting worst behavior disabling Z-hop. Still need to test with the Ender-3. Could you share all your combing related settings ? Such as part avoidance and travel distances. I'm probably going to disable Z-hop and enable combing infill by default, adding a simple rule to adjust combing (infill/off) depending on Z-hop (disabled/enabled).

I'll make a full report with pictures by the end of the week with a new beta to try out..

trouch, 2019-06-05:
Forgot the Infill.
- Lowering Infill to 10% with Octet or Cubic makes small features such as the Calicat front feet not being supported.

PriestWithKnives, 2019-06-05:
For combing settings I tend to find these settings work well

https://cdn.discordapp.com/attachments/414502838266494988/585957763377790996/unknown.png

 Odd that the octet infill is causing an issue with calicat, might be worthwhile to try gyroid, it's a pretty decent cross of zigzag and smooth curves.

trouch, 2019-06-06:
Here is a new def to try that takes in consideration some of your suggestions.
https://github.com/trouch/CreawsomeMod/blob/beta/4/src/definitions/creawsome_base.def.json

To be honest, I'm less confident in overall quality with that one compared to my original, but it should fix strings, gaps and details loss issues to people who encounter it. 

I still think that disabling Z-hop is only a workaround to proper calibration and overheating issues. Disabling Z-hop brings scares on bottom/top surfaces. Also combing with model and support avoidance makes a lot of X/Y moves and extra vibrations. Eventually, on a multi model print, we'll have vertical scares where the nozzle crosses perimeters. Since I don't have any string issue - I had to tweak my settings to get strings, which is ridiculous - my personal preference goes for Z-hop enabled, but I disabled it by default.

Finding the right line width is really a pain in a the *** to fit any model tolerance. I ended with a 110% as a tradeoff. No matter of the settings, there is a good chance of having either gaps or thin infill instead of wall and dimension accuracy issues.

trouch, 2019-06-06:
_Adding few pics..._

**Heating effect on strings with an Ender-3 inside enclosure with Z-Hop enabled. Keeping the door closed is a bad idea when printing PLA.**
[image]

**Vertical scare when Z-Hop is disabled on Ender-3 within enclosure.**
[image]

**CR-10S Pro Quick comparison. The cat tail has weird defect when Z-hop is disabled.**
[image]

**The Calicat army.**
[image]

PriestWithKnives, 2019-06-06:
Were those with the combing I listed above on, also yeah without the z hop definitely less stringing in your string test. Might be time to try Gyroid at say 14%. It has decent strength while also causing less vibration as it needs less fast movements in comparison to something like cubic. Pretty strong too.

[image]

While not taking quite as much time as some others

https://i.stack.imgur.com/BDfQW.png

PriestWithKnives, 2019-06-06:
Looking at the new def, not sure about the 65 degree angle on supports, past about 45-50 you get into trouble without really good cooling.

support_angle": { "value": 65 },

trouch, 2019-06-06:
It has always been set to 65.
funny fact someone asked to increase to 75 in #67 which I refused.

PriestWithKnives, 2019-06-06:
Yeah got to be honest, anything past 50 and you're asking for angry people asking why their supports aren't working and they're having drooping pla. It's defaulted to 45 in S3D for a reason. Now I know people with dual 5015 setups who can blow past to 70 no big deal, but the majority of people are using stock or maybe a petsfang bullseye setup with the oem fan or 1 5015, with one of the latter you can get a bit past 50 but even then you're pushing it.

PriestWithKnives, 2019-06-06:
Example
https://cdn.thingiverse.com/renders/cd/1a/b5/13/28/2e363501147c43f06febaeaa0c40d9dc_preview_featured.jpg
Just looking through thingiverse, with comment
"All tests passed, but a few didn't do as well as I would have liked.
The underside of the taller overhang test starting at 50° becomes very rough, much rougher than the shorter overhang test. It affected the finish of the sides.
The underside of the shorter overhang test starting at 60° becomes a bit rough, but not bad enough to affect the sides."
and this Ender 3 Pro
https://cdn.thingiverse.com/renders/de/7b/29/69/6c/824551ca47837290ad9124dcc9756b0d_preview_featured.jpeg
You can see it starting to deform around the 50 mark.

trouch, 2019-06-07:
To be honest, what ever I decide, I'll have to deal with angry people because the configuration doesn't work out of the box for them or because it's not finely tuned for their customized printed.
I can't make a 100% compatibility profile unless I provide everything - firmware - extruder - hotend - nozzle - cooler - filament .... hence Prusa and Ultimaker business 😅 

With regards to the 65° angle, I've made a compromise as it works with stock cooler when using Super / Ultra Qualities.

PriestWithKnives, 2019-06-07:
So it works when going very slow in extremely small layer heights? Oh boy...at least set it to 50. 65 is a really too much for stock cooling, .12 will have issues as will .16 and above. I know you're trying to cut time but here you're cutting quality.

trouch, 2019-06-24:
I've came down to a better generic solution by computing maximum angle which depends on layer height and line width. We go up to the half of a line width in the air before adding support, which come down to a simple reverse tangente formula. And if you try it with 0.4 line width and 0.2 layer heigh, you get usual 45° angle. Then lowering height or increasing width allows for bigger angle. Et voilà !

So far beta 4 is almost ready and base definition is available in its own branch. Hopefully it will fix issues people encounter.

Keichide, 2019-07-13:
I would also prupose a little tweak.. 
Set Top Surface Skin Layers on 2, set the Top Surface Skin Pattern to Lines and set the Top/Bottom Pattern to concentric, this way it prints a little bit faster but without the usual little dots on the beginning and the end of each line pattern move, which is probaly a level issue but anyway it's something you usual does'nt see but the print profits of it.

trouch, 2019-07-16:
Mod is merged upstream, I'm closing this issue as Cura 4.2 is on the corner.
Please report new issue here for any improvement request with Cura 4.2 bundled mod.

closed 2019-07-16 by trouch