trouch/CreawsomeMod #73: Supports Difficult to Remove

agrias, 2019-06-03:
Using Cura 4.0 and printing with supports even with 5% support density is super hard without breaking the print. Any advice/guidance would be appreciated

trouch, 2019-06-04:
What version of the mod and what quality / layer height are used ?

agrias, 2019-06-04:
CreawsomeMod-0.2.zip , Standard / .2

Will try with Creawsome3

trouch, 2019-06-06:
interface has been fixed with beta 3. please report if it still happening.

closed 2019-06-06 by trouch

agrias, 2019-06-11:
@trouch Even with beta 3 the supports are still extremely difficult to remove. I did a new installation of 3.0 and had no better luck

Al3601, 2019-06-11:
@agrias i solved this problem by applying this settings (tested for PLA & PETG) (creawsome Beta3.2 + cura 4.1.0)
[image]

hope this can help you, by the way is was the only problem i was facing with Creawsome

RFC1795, 2019-06-15:
I'm having the same problem. Even with those suggested settings @Al3601 ... a bit better, but I couldn't get the support off, even using a long nose pliers :(