trouch/CreawsomeMod #75: Request for PVA/HIPS and Second Nozzle option

TheSin-, 2019-06-04:
Firstly thank you for all your hard work this mod is amazing and made Cura viable for me, when I went form 1 nozzle with 2 so I could print PVA supports I lost Slic3r since it doesn't have prime towers.  And Cura prints where horrible.

Currently I have to add an extruders to the base, copy your extruder_0 to extruder_1 and then mod a bunch of things to enable HIPS and PVA so my 2nd extruder can do supports.  I'm not done testing yet as I hit a clogged and wanted to fix it first (nozzle wasn't flush on the heat break).

Anyhow this isn't crucial, just nice if I didn't have to remember all my changes on every update.  My nozzle is a NF TC-02 so it's a 2 in 1 so the nozzles have no offset making it very easy, I just have to set my extruder swap to a 35mm retract and I'm gtg.

If you'd like a PR of my current setup I'd be happy to provide them but they are ender3 only currently and 0.4mm only. I'm a novice still so I'm sure your understanding of the options and settings is much better at it them mine, I left the defaults to 1 extruder cause I didn't want to change things too much.

I'm also working on a Generic TPE profile that I'd love to contribute once I have it printing nicely.

Warsenius, 2019-06-07:
I have modded my ender to have 2 extruders usign a y-joiner. it would be great if the number of extruders is editable in the creawesomemod.

joaomamede, 2019-08-19:
+1 on this. I have a ender 3  pro with two extruders and I need to add custom printer and then I can't use creawsome mod. 
Cura does everything for us, it's just that if forces only 1 nozzle/extruder when configuring ender 3.
Is there any way to do this? 
TheSin, what things did you change (which files specifically?)

TheSin-, 2019-08-19:
I'm currently working on this with Cura 4.2.1 which Creawesome is now built in, I'll let you know once I have it sorted. The big things is adding a second extruder and at the time I had to mod the printer definition and copy the extruder_0 file to extruder_1, then I could select a 2nd one, then it was just a matter of added the gcode for extruder switching in the config.  The harder part was getting HIPS and PVA support.

joaomamede, 2019-09-04:
TheSin can you help me into using configuring the files for ender 3 in cura 4.2.1? I don't know where to start without messing it up.

trouch, 2020-03-21:
please submit PR upstream.

closed 2020-03-21 by trouch