trouch/CreawsomeMod #79: Crash Cura 4.1 when trying to add non-Creawsome Ender-3

jpcuzzourt, 2019-06-10:
I'm using Cura 4.1 and the matching Creawsome Beta 3.2 release. 
Windows 10 64-bit.

I just ran into a crash that is repeatable, for me at least.
I can add additional printers of various types, like Ultimaker S5, and Creality CR-10 S4.
But if I attempt to add a standard Creality Ender-3 printer, Cura 4.1 crashes.

I do have a Creawsome Ender-3 printer, and I've tried renaming it, and also the new printer that I try to create so that there is not a printer name clash, but that doesn't make a difference.
Here is the "Error Traceback"

	Traceback (most recent call last):
	  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\MachineManager.py", line 404, in addMachine
	  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 68, in createMachine
	  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 133, in createExtruderStackWithDefaultSetup
	  File "X:\4.1\build\inst\lib\python3.5\site-packages\cura\Settings\CuraStackBuilder.py", line 129, in createExtruderStackWithDefaultSetup
	IndexError: list index out of range

It is possible that this is some corruption of my own profile from a previous Ender-3 configuration that I did not completely get cleared up.
After trying to restore from a 4.0 backup of my profiles, I ran into some "corrupt profile" errors and had to reset my configuration, but I thought that had cleaned everything up.
So if you aren't able to repeat it, it's probably just me.
Don't waste too much time or effort looking into this!
I'm trying to upload my cura.log, but it isn't completing for an unknown reason (Maybe just too big.)

If there's an easy fix for this on my end, I'd like to know though.

zebons, 2019-06-11:
Works fine for me : Cura 4.1 with creawsomemod addon and I can add a standard Creality Ender-3 printer

trouch, 2019-06-25:
Delete the cura.log, start over to reproduce issue, and zip the log file then.
Starting with a clean setup shouldn't bring any issue.

closed 2019-07-16 by trouch