trouch/CreawsomeMod #81: pas double extruder dans "custome machine"

prepajar, 2019-06-11:
Bonjour,
Tous d’abord bravo pour ce travail !! j'ai test sur une ender 3 et c'est parfait !!!
En revanche, je voulais tester sur une Anet A8-m à doubles extruder et je constate qu'il n'y a pas la possibilité de configurer en "custome machine" avec double extruder. J'ai pensé que c’était un oubli pour la version 4.1 de CURA mais même problème sur 4.0 CURA. Avez-vous en projet de rajouter ce manque ?  
     Hello,
First of all congratulations for this work !! I test on an ender 3 and it's perfect!
On the other hand, I wanted to test on an Anet A8-m to double extrude and I see that there is no possibility to configure in 'custome machine' with double extrude. I thought it was an oversight for version 4.1 of CURA but same problem on 4.0 CURA. Do you plan to add this lack?
Merci

trouch, 2019-06-25:
Yeah, but I need to play with dual and multi material first !
That's a low priority for now.

trouch, 2019-06-25:
Et merci ;-)

trouch, 2019-06-25:
https://github.com/trouch/CreawsomeMod/issues/8

closed 2020-03-21 by trouch