trouch/CreawsomeMod #83: Cura 4.1 Crash

nvladik, 2019-06-15:
Crash on a brand new installation of cura (so no settings to clean up). I installed Cura 4.1, backed up Resources folder, copied over the new folder, as soon as Cura tries to start there's a crash "application quit unexpectedly" with nothing useful in the crash report.

infrat, 2019-06-16:
Are you sure you'd replaced correct resources folder? Keep in mind, that there are two 'Resources' folders inside Cura package. One is on the highest level (started with BIG letter), and the second one inside (started with small letter). You need to replace that second one.
[image]
 It might be little bit confusing (so my advice for Author is to update Readme with that note)

trouch, 2019-06-25:
most of crashes are caused either by 
- wrong zip file downloaded, you have to download it from the releases tab
- wrong `resources` folder replaced, need to replace the lower `r` one
- user setting corruption upon existing profil import, which a settings reset solves

closed 2019-06-25 by trouch