trouch/CreawsomeMod #86: Excessive retractions while printing infill keep jamming up my extruder gears

midicdj1000, 2019-06-24:
I have stock cr20-pro that seams ok, but on my ender 3 with upgraded ALU extruder gear it will randomly stop extruding and you can see the filament is very thin where the gear has ground it down. I notices the retraction is on all the time while doing infill so this is slowing down my prints by perhaps 33%, but now both prints using 4.1 profiles failed with this fault of no extruding because the gear ground the filament, If I go back to my old 3.6 profile it will print without issue. Now I actually like the quality of the new profile especially the first layer when it has lots of details like small circles for screw ect so I do not want to get rid of it, just reduce it on parts like infil. Is this possible ? I looked to see if I can have one profile for the first layer and then change to cut back after, but this option was very basic on a plugin I could find.

trouch, 2019-06-24:
lower down retract speed to 25mm/s for testing purpose.
if that works, then you have clogging upon high speed retract because of very slight and unnoticeable leak between the PTFE and nozzle.

midicdj1000, 2019-06-24:
I will try it, but can i reduce the sensitivity to stop it retracting while doing the infill ? it makes printing so slow.

midicdj1000, 2019-06-24:
[image]

midicdj1000, 2019-06-25:
Just doing a comparison between my 3.6 profile for a part I print in a kit I make so time is a big factor and it's 69 mins, compare this to the 4.1 creawsome profile of 101 mins (97 if I tweak it to match the exact settings of 3.6)  and this is not something I could work with. As nice as the prints are, taking this much extra time would make a big impact for me.

trouch, 2019-06-25:
Your picture looks to be caused - as per my experience - by a clogged nozzle occurring with high speed retract. I was expecting a test with lowered retract speed to confirm that. Given the settings of the mod, it requires very well set extruder.
If you read the doc you will see your printer limits retract speed to 25mm/s what ever you set in Cura without the mod, hence you don't have any issue without the mod.

The print time comparison does not make sense and doesn't help to solve that issue. But up to you to choose between fast print or quality print, that's it.

closed 2019-06-25 by trouch

midicdj1000, 2019-06-25:
I have to print some parts  for kits so will not have time to test the 25mm/s , but I will do so soon. In this case quality is not that important, but I think you would agree the time difference is quite a lot for the same print.

trouch, 2019-06-25:
I did NOT ask to set PRINT speed to 25mm/s. 
I asked to set RETRACT speed to 25mm/s, in the material settings 😬 
At least it confirm you did not read the doc.

midicdj1000, 2019-06-25:
wow hang on a moment who said print speed ? I never mentioned anything apart from 25mm/s so you have assumed I was talking print speed when I never did. This is not very helpful when you instantly have an attitude  after I said I would do what you asked at a later date.