trouch/CreawsomeMod #88: CreawsomeMod-4.1.0 Z Seam Alignment - Sharpest Corner don't work

juckendesAuge, 2019-06-29:
Hello,
Cura doesn't always start at a corner with CreawsomeMod-4.1.0.
I am currently printing https://www.thingiverse.com/thing:73489
penholder.stl
04-14-13
82927 29mb

Starting point 1 is ok.
But at 2 it starts in a curve.

[image]

Many greetings

mkh97, 2019-07-15:
[image]
Try This

juckendesAuge, 2019-07-15:
I have test user Specified.
The seam is then in the middle of the printed object.

[image]

Or is Z seam y 90 decisive?

Thank you very much for help

mkh97, 2019-07-15:
I think try both a minimal same parts and give your decision but my idea is y 90

juckendesAuge, 2019-07-15:
Thank you very much.

trouch, 2019-07-16:
The Y=90 is just an example to show how you can move to a corner

Seam corner preference is only effective when there is a corner on the layer. Totally curved parts don't have any corner.

With regards to the pig, the seam splits in two where you can't have a single perimeter. The ears are basically two separated circles at this point.

Cura just tries to place the seam as close as possible to the X/Y positions configured and to fulfill corner preference. Hence I set it to "None" to only use X/Y positions.

closed 2019-07-16 by trouch