trouch/CreawsomeMod #90: Slightly adjust start gcode

Keichide, 2019-07-13:
I did'nt like your start gcode (and also never liked the one from Ultimaker) since it was always pulling a string while homing, so i modified a bit, to heat up the nozzle just to 140 degree, which is rigt before the melting point of most materials, and home first before heating up all the way to the final temperature. This way it avoids the string on the build plate (or the nozzle).

M201 X500.00 Y500.00 Z100.00 E5000.00 ;Setup machine max acceleration
M203 X500.00 Y500.00 Z10.00 E50.00 ;Setup machine max feedrate
M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate
M104 S140; start warming extruder to 140
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
G28 ; Home all axes
M420 S1 ; Enable Bed Level Mesh (optimal)
G1 X0.4 Y20 Z0.3 F5000.0 ; Move to start position
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G92 E0 ;Reset Extruder
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
G1 X0.4 Y200.0 Z0.24 F1500.0 E15 ; Draw the first line
G1 X0.6 Y200.0 Z0.24 F5000.0 ; Move to side a little
G1 X0.6 Y20 Z0.24 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed

Oh and i reversed your prime line to the edge of a 235x235 Bed again, because it was way to much into the print area.

For the feedrate and acceleration settings... We will see how the turn out, since i already set different ones directly in the marlin configuration.h

ipatch, 2019-07-16:
just tested your start gcode steps and so far so good.  i like how you get the hot bed to temp before setting the nozzle temp and then placing the nozzle close to the bed so that the nozzle burgers are kept to a minimum while the nozzle is heating up to temp.

Keichide, 2019-07-16:
Thank you. Like i said the best thing about it, is that the Bed stays Clean all the Time, since the Hotend is just going until 140C and slowly heats up to the melting Point (and final Set Temp in the Profil) while the Printer does the Homeing, which is still enough time that it won't ooze but also fast enough that he can Start the Print after the Homing.

Also i prefer the Prime Line on the very Edge of the Build Plate thats one of the good things the guys from Ultimaker implanted. Since i can easily remove it while he prints.The only thing i'm still thinking about is a adding a Retract but thats probably overkill

rap66, 2019-07-29:
Keichide
A little late to the discussion but I'm still totally green when it comes to this Gcode/Script stuff. Is your script a total change to the original creawesome scripts??  In this case it would be a copy & replace?? or would I just add it to the already existing Script? I'm also using TH3Ds EZABL - which has some Code you need to add as well. Is there a specific order this code needs to be?
Thanks
Rich

Keichide, 2019-07-30:
It's Copy & Replace since it also changes the Location of the Prime Line.

rap66, 2019-07-31:
OK! I'll give it a try 
Thanks!!

trouch, 2020-03-21:
please submit PR upstream eventually.

closed 2020-03-21 by trouch