trouch/CreawsomeMod #91: Z Offset Ignored on CR-20 Pro

rjnich, 2019-07-15:
I am using the basic CR-20 Pro profile from the resources folder.  When printing, I receive a transparent first layer after the initial Cura purge line results without an issue.  Actual extrusion is only noticeable on following layers.  Z-Offset on the machine is set to -3.00 and prior to using the mod did not have an issue.   After testing different offsets, I have noticed no changes while using the profile or any modification of the starting gcode.
[image]

trouch, 2019-07-16:
change the M420 with a G29 within the start gcode and report

rjnich, 2019-07-17:
That seems to have fixed the issue, thank you for the assistance.

unimonkey1994, 2019-07-29:
Ok.  So I tried this but it didn't work as expected.  what is it just "G29 Z0" or something else at that line?

closed 2020-03-21 by trouch