trouch/CreawsomeMod #92: TPU material not supported?

isometriq, 2019-07-18:
I've set up a new material for my TPU filament. The the model was grayed out and I could read that it was not supported. To print, I've putted another material type PETG to fool the system and then I was able to print.

I used the `Dynamic Quality` preset and disabled `Retraction`. It did work wonderfully on first try ..so why it is marked as not supported. Am I lucky?

Printer: Creality Ender 3 with glass bed
Dynamic preset
Retraction: Off
Filament: 3D Printing Canada TPU
Bed temp: 230
Nozzle temp: 80

tvixen, 2019-07-22:
Bed temp: 230? 
Nozzle temp: 80?

Are U sure of that?

isometriq, 2019-07-22:
Yes absolutely and that's middle in the recommended range, which is:
Bed: 75 to 85
Nozzle: 220 to 240

There is some stringing in very specific spots (same x/y location), but it is easy to cut and clean (it's like combed hair).

tvixen, 2019-07-22:
Well, thats not what you was writing. Thats why I was asking.  Read your own text.
The stringing is probably because the air from your blower is making turbulence from other buildups around the plate.

isometriq, 2019-07-23:
Apart the slight stringing which was very easy to remove (not a huge mess), the result was very good. I was able to print a few functional things and very happy of the result.

My point is that from my current experience, I think TPU should be considered as supported. I should not have to use another material type name to be able to print it. Such restriction is not useful, especially if one upgrade its printer to handle a "non-supported" or in my case when it just works. Maybe a warning would be more useful.

tvixen, 2019-07-25:
I can't say that I have the same problem, as I did what the instructions said. Copy all materials and make your own. No problems with TPU.  At all.

isometriq, 2019-07-25:
I was trying to grab a screenshot to show the symptom (slice button was grayed with invalid message), but now it seems to work by changing the material back to TPU.. oh well, something bizarre must've happened...

In the material section, it says it is linked to the PETG material, not sure why and what that means...
[image]

EDIT:
So I've unlinked it and now if I select the TPU material, the dropdown with the profiles provided by Creawesome shows "Not supported".
[image]

Thankfully, I've already created a profile for TPU specific tweaks based on "Dynamic Quality" profile and saved it as a profile. I'm able to use this one. But by default, it is confusing. Seems like the profile have specific material compatibility, which makes sense, but very confusing if you want to extend a Creawesome profile (which are the only ones provided when mod is applied to Cura).

isometriq, 2019-07-25:
And my original TPU material, was an duplicate of the PETG material, but the actual "link" seem to have happened when I specified the type to be PETG.

tvixen, 2019-07-25:
I would re-install Cura, and the Creawsome Mod again. Apparently something went wrong.

trouch, 2020-03-21:
need further test and modification.

closed 2020-03-21 by trouch