trouch/CreawsomeMod #96: Can't Slice when using support in cura 4.2

sihyeonlee, 2019-07-30:
Hello, I have a problem with my experience when using cura 4.2

When I want to use a support option, that has an error.

"Unable to slice - with the Z Hop Speed"

So I changed Z Hop Speed 10 to 5, but it doesn't resolve the error.

Printer - CR-10S Pro
Profile - 0.4mm nozzle Standard Quality
Cura Version - 4.2

nallath, 2019-07-30:
I've just tried it and it works for me (after I set it to 5 that is)

sihyeonlee, 2019-07-31:
@nallath I set that up same, but I still get a slice error.
Can I know how set up your Profiles? Now I use Standard Quality profile.

sihyeonlee, 2019-07-31:
I solved the problem, and it problem was in Cura 4.2!!!

==Solution==
1. Go to Cura MarketPlace and install Printer Settings.
2. Preferences > Configure Cura > Settings > Printer Settings
3. Check the 'Maximum Speed' Z Combo box
4. Return to Home and Edit Print Settings > 'Maximum Speed Z : 10'

closed 2019-07-31 by sihyeonlee

nallath, 2019-07-31:
As I posted in a few other issues, there were two things going on. One of them was a weird setting (max speed Z being set at 5 and zhop set at 10). The other was an issue with our error checker which only occured in very specific cases (the error has been there for a long time, but the specific case only popped up now).

We're working on a patch release right now. The code is already pushed, but it's being tested now. You can expect a  4.2.1 release today / tomorrow if all goes well.

clumsyzombie, 2019-07-31:
Do you know if they are going to be updating this mod for Cura 4.2.x?