trouch/CreawsomeMod #97: Ender 3 dimensions

batrappo, 2019-07-30:
I loaded the ender3 profile from creawsomemod list but I found wrong dimensions! Why? Ender 3 is 235x235 while in your profile is 220x220, is there a reason or is it a mistake?
Thanks in advance

randellhodges, 2019-07-31:
While the bed is 235x235 and that is what people end up using, the official, published print area is only 220x220.

Probably just being safe and using the published print area.

batrappo, 2019-07-31:
thanks for the answer...I was thinking the same but if you choose ender 3 from the printers list in "non mod" CURA 4.1 you will see that the dimension is 235x235...because of this difference I had doubt...

trouch, 2019-08-08:
that's a 50/50 issue :-) 
Some people prefer to have "real" dimension of the buildplate.
Some people prefer to have "specified" dimension which is safer.

You can still change buildplate dimension on your own if you need.

closed 2019-08-08 by trouch