trouch/CreawsomeMod #98: Cura 4.2?

clumsyzombie, 2019-07-31:
Dose anyone know id they are working on a update for cura 4.2?

nallath, 2019-08-01:
The profiles have been merged into 4.2, so there isn't really a need to install them by hand.

ll777, 2019-08-01:
I Couldn't find them in 4.2.1 where should I look for?

remy561, 2019-08-01:
@ll777, they have been merged into the Cura default settings, so you should use the default Ender 3 profile. See the release notes of v4.2: https://github.com/Ultimaker/Cura/releases

EvilSpock, 2019-08-02:
So what is the process to go from Creawesome 4.1 to Cura 4.2.1 with no errors.  Do we do a restore to remove Creawesome before upgrading to Cura 4.2.1 or just run the upgrade to 4.2.1.  Also is the look at feel with the machine views or do we loose the Creawesome look.
Thanks

trouch, 2019-08-08:
No more mod with Cura 4.2+
All the mods settings are now included out of the box.
The only exception is the Ender-3 preview - model was too detailed and too big to be integrated.