trouch/CreawsomeMod #112: No machine preview

DarkDudae, 2019-10-22:
Since the profiles have been integrated into official Cura versions, the feature of the preview 3d machine is not there anymore. I have an old resources folder downloaded but I don´t know what files should I restore to recover that interesting feature. Any clue?

pearlsails, 2019-10-31:
i too miss this feature, felt very fancy. I don't know why they switched to this, the old way of installing wasn't an issue

nallath, 2019-11-01:
Because it was wreaking havoc on our crash reports. A lot of people were using this, and also installing wrong versions of the mod with the wrong versions of Cura.

Anyway, the models can be added, someone just needs to make a pull request and ensure the models are not too big.

Ithallius, 2020-03-18:
So, you can use the mesh that was in Creawesome (At least from 4.1) in 4.5 I've found, by renaming it to creality_ender3.stl and copying it into Cura's meshes folder, replacing the stl that is there. However, the model is off-center. Looking at the mod's definition json, it had the following line in it to account for this:
"platform_offset": [ -160, 0, 140]
However, copy and pasting that into Cura's definition json for the Ender 3 causes the profile to become corrupt as far as Cura is concerned. If anyone knows how to fix this, I think it'd solve the requests for the 3d printer preview can be addressed.

trouch, 2020-03-21:
STL was too big and Ultimaker requested to remove it.

closed 2020-03-21 by trouch