trouch/CreawsomeMod #113: Support generation

frozenPG, 2019-11-09:
This mode elevates the quality to a great level. Unfortunately, it messes up the support generation :( Since it was hard coded into Cura 4.2 and newer, it is quite a problem. Even messes with support interface. See the screens.
[image]
[image]
[image]

frozenPG, 2019-11-09:
it should look like this (from working 4.1):
[image]

pearlsails, 2019-11-10:
Oh, this is a nightmare, the 4.3 version is so bad compared to the older one for 4.1. Walls have issues fusing together, support generation is awful. I've replaced the hotend and calibrated esteps and still issues. Wish we still had the option to get the old one as i cant get it back now

frozenPG, 2019-11-13:
OK, I made some progress. It seems, that from reasons unknown Crewasome mod (hardcoded in Cura >4.2), changed DEFAULT settings for supports! So if you come with your previously flowless support settings from previous versions, it will simply fail. Your profile contains only changes (what is not changed is taken from default Cura settings WHICH HAS CHANGED!). Even the standard profiles in Cura will fail at creating reasonable supports! So, what is the solution? Simple. Open your working profile in Cura 4.1, show ALL settings, make notes (screens) from the support section and then compare it with Cura 4.3 - change accordingly. Done :)

closed 2019-11-13 by frozenPG